bblanchon/ArduinoJson #2046: v7: Stack usage during serialization is significantly higher than v6

willmmiles, 2024-02-01:
**Describe the bug**  

Serialization using ArduinoJSON v7 on ESP8266 seems to be using substantially more stack than v6 for common serialization tasks.  With a large configuration structure, this can result in a stack overflow, causing difficult to diagnose crashes.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v7/troubleshooter/):  
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns serialization
4. Program crashes
5. Program uses `PROGMEM`
6. Casting the pointer doesn't solve the issue

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266
* Core/runtime: ESP8266 core for Arduino v3.1.2  (PlatformIO espressif8266@4.2.1)
* IDE: VSCode/PlatformIO

**Reproduction**  

[TODO] Working on it - I should be able to provide a better (complete and buildable) example in a few days.   Serialization code is of the form:

```c++
void serializeConfig() {
  JsonDocument doc;
  JsonObject root = doc.to<JsonObject>();

  JsonArray rev = root[F("rev")].to<JsonArray>();
  rev.add(1); //major settings revision
  rev.add(0); //minor settings revision

  root[F("vid")] = VERSION;

  JsonObject id = root[F("id")].to<JsonObject>();
  id[F("mdns")] = cmDNS;
  id[F("name")] = serverDescription;
  id[F("sui")] = simplifiedUI;
  // and so on ...

  File f = LittleFS.open("/cfg-v7.json", "w");
  if (f) serializeJson(root, f);
  f.close();
}

void measureStackUsage() {
    cont_check(g_pcont);   // Validate we haven't already overflowed
    cont_repaint_stack(g_pcont);  // Fill unused stack with guard value

    auto unused_stack_before = cont_get_free_stack(g_pcont);    
    serializeConfig();
    auto unused_stack_after = cont_get_free_stack(g_pcont);

    Serial.printf("Stack used by serializeConfig: %d\n", unused_stack_after - unused_stack_before);
}
```

willmmiles, 2024-02-01:
I tried using ghidra to inspect the stack frames.  It looked to me that the difference compared to v6 seems to be a combination of both larger sizes for JsonObject/JsonArray, and a plethora of FlashString objects left on the stack.   It seems gcc isn't able to recognize that the FlashStrings are temporary and the stack can be recovered.  Passing -fstack-reuse=all did not help.

willmmiles, 2024-02-01:
As a workaround on ESP8266es, increasing the stack size with `-D CONT_STACKSIZE=8192` worked around the crashes in my project, though this reduces the stack size available to the system context and it's unclear if that will cause trouble with system functionality later.  RTOS-style runtimes with small task stacks (such as ESP32s) might also be impacted.

bblanchon, 2024-02-01:
Hi @willmmiles,

Thank you very much for reporting this issue.

I turn your repro code into the following program:

```c++
#include <Arduino.h>
#include <ArduinoJson.h>
#include <cont.h>

void serializeConfig() {
  DynamicJsonDocument doc(1024);

  JsonArray rev = doc[F("rev")].to<JsonArray>();
  rev.add(1);
  rev.add(0);

  doc[F("vid")] = ARDUINOJSON_VERSION;

  JsonObject id = doc[F("id")].to<JsonObject>();
  id[F("mdns")] = "cmDNS";
  id[F("name")] = "serverDescription";
  id[F("sui")] = "simplifiedUI";

  serializeJson(doc, Serial);
}

void setup() {
  Serial.begin(9600);
  while (!Serial)
    continue;

  cont_check(g_pcont);         // Validate we haven't already overflowed
  cont_repaint_stack(g_pcont); // Fill unused stack with guard value

  auto unused_stack_before =
      cont_get_free_stack(g_pcont); // Counts stack filled with guard value
  serializeConfig();

  cont_check(g_pcont); // Validate we haven't overflowed
  auto unused_stack_after = cont_get_free_stack(g_pcont);

  Serial.printf("Stack used by serializeConfig: %d\n",
                unused_stack_before - unused_stack_after);
}

void loop() {}
```

With it, I got the following results:

| ArduinoJson | Arduino IDE | PlatformIO Release | PlatformIO Debug |
|-------------|-------------|--------------------|------------------|
| v6.21.5     | 532         | 500                | 960              |
| v7.0.2      | 580         | 548                | 1184             |
| c98b05e2    | 580         | 548                | 1120             |

We can see an increase, but it's not that big.

I also did some tests by modifying the library to print the stack pointer address and got the following results for six levels of nesting:

| ArduinoJson | Arduino IDE | PlatformIO Release | PlatformIO Debug |
|-------------|-------------|--------------------|------------------|
| v6.21.5     | 256         | 240                | 688              |
| v7.0.2      | 368         | 256                | 1312             |
| c98b05e2    | 256         | 192                | 1008             |

We can see a significant increase, but only in debug mode.
Are you having problems in debug or release?

c98b05e2 corresponds to the current head of branch `7.x`.
Can you give it a try?

Best regards,
Benoit

willmmiles, 2024-02-01:
Thanks for taking such a detailed look, I really appreciate it.

I'm using release builds, and unfortunately c98b05e made no significant difference for the whole config.  I finally instrumented my project properly using the logic from above, with the full config, and got the following results:

| ArduinoJson | Stack Usage |
| --------------|---------------|
| v6.21.5         | 2308 |
| v7.0.2           | 4948 |
| c98b05e       | 4804 |

Possibly there's some kind of phase change in the compiler output if the function gets too big (eg. the optimizer times out and gives up partway, or something).  Here is a copy of the whole cfg, for reference:  [cfg-tmp.json](https://github.com/bblanchon/ArduinoJson/files/14126698/cfg-tmp.json), and a link to the generating code, in case you can spot some obvious usage error: https://github.com/willmmiles/WLED/blob/c2d2000f9b91fcb02fd37b7f255911ff4fe5b5ac/wled00/cfg.cpp#L640

willmmiles, 2024-02-01:
The disassembly analysis suggested that the compiler is also storing all the temporaries at unique stack addresses, too -- every object looked up with `[]`, and (with v7) a `FlashString` for every `F()`.  I don't know why it's refusing to re-use the same stack for each one.  As a quick test, maybe you could try adding more data members in your test program to see if it increases the stack usage for each one?  If you don't see any increase in utilization, that narrows it down to something unique to my build configuration, so I know where to start looking next.

I can definitely work around the issue by breaking up the large configuration to smaller functions (assuming the compiler doesn't undo it all by inlining them!).  I'm not quite ready to give up yet, though - I'd hate to be the guy debugging some project that suddenly got unstable when I added one more entry to the configuration file, or to have to document "you must break up large serializations in to small functions because otherwise it might crash on some systems".

willmmiles, 2024-02-02:
You were on the right track with looking at inlining.  It looks like gcc (at least the version in this toolchain) isn't able to share space in the stack frame for inlined functions -- every inlined copy has to be allocated unique stack space for their own arguments.  This is the cause of the additional stack usage.

As an experiment, I tried globally disabling FORCE_INLINE in ArduinoJson, in favor of letting the compiler decide when to inline.  The results was a large reduction in stack usage (down to 1572 bytes!!),  a 14528 byte reduction in program size, and no noticable performance change for my use case.  If it's helpful, the compiled `serializeConfig` function became basically a list of `detail::VariantRefBase<>::to<>`. `JsonArray::add<>` and `detail::VariantRefBase<>::set<>` calls -- these might be the right places to not force inlining.

bblanchon, 2024-02-02:
That's the opposite of my experience with `FORCE_INLINE`: it often helps me reduce the code size and stack usage.
Can you pinpoint the ones that cause this large increase?

bblanchon, 2024-02-02:
I did some scripting to find which `FORCE_INLINE`s were actually useful.

I compiled in the following table the impact of each on the size of ArduinoJson's examples: `JsonParserExample`, `JsonGeneratorExample`, `MsgPackExample`, `StringExample`, and `ProgmemExample`.
I used AVR as the target CPU because it's where code size is usually an issue, and because it compile quickly.

File:line                        | Parser | Generator | MsgPack | String | Progmem
---------------------------------|--------|-----------|---------|--------|--------
Array/ElementProxy.hpp:31        | 0      | 0         | 0       | 0      | 0
Array/ElementProxy.hpp:37        | 0      | 0         | 0       | 0      | 0
Array/ElementProxy.hpp:43        | 0      | 0         | 0       | 0      | 0
Array/ElementProxy.hpp:47        | -40    | 0         | -66     | 0      | 0
Array/ElementProxy.hpp:53        | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:23           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:26           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:66           | 0      | -72       | 0       | 0      | 0
Array/JsonArray.hpp:73           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:79           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:87           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:93           | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:108          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:114          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:126          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:136          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:142          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:148          | 0      | 0         | 0       | 0      | 0
Array/JsonArray.hpp:154          | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:26      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:34      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:39      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:42      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:48      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:59      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:65      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:71      | 0      | 0         | 0       | 0      | 0
Array/JsonArrayConst.hpp:77      | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:173    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:183    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:193    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:203    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:212    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:218    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:243    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:250    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:256    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:263    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:272    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:279    | 0      | 0         | 0       | 0      | 0
Document/JsonDocument.hpp:283    | 0      | 0         | 0       | 0      | 0
Json/Latch.hpp:28                | -64    | 0         | 0       | -70    | -66
Object/JsonObject.hpp:23         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:26         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:46         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:52         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:58         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:64         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:70         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:78         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:90         | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:106        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:116        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:125        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:132        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:140        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:148        | 0      | 0         | 0       | 0      | 0
Object/JsonObject.hpp:158        | 0      | 0         | 0       | 0      | 0
Object/JsonObjectConst.hpp:35    | 0      | 0         | 0       | 0      | 0
Object/JsonObjectConst.hpp:41    | 0      | 0         | 0       | 0      | 0
Object/JsonObjectConst.hpp:47    | 0      | 0         | 0       | 0      | 0
Object/JsonObjectConst.hpp:53    | 0      | 0         | 0       | 0      | 0
Object/JsonObjectConst.hpp:59    | 0      | 0         | 0       | 0      | 0
[... 37 lines cut ...]
Variant/VariantRefBase.hpp:94    | 0      | 0         | 0       | 0      | 0
[... cut ...]
Variant/VariantRefBase.hpp:265   | 0      | -48       | 0       | 92     | -40
Variant/VariantRefBase.hpp:269   | 0      | 0         | 0       | 0      | 0
Variant/VariantRefBase.hpp:271   | -24    | 0         | -202    | 74     | 0
Variant/VariantRefBase.hpp:275   | 0      | 0         | 0       | 86     | -36

In this table, a negative number means the size decreases (good) and a positive number means that the size increases (bad).

I already remove the 103 `FORCE_INLINE`s that didn't affect the code size, so make sure you pull the lastest version from the `v7` branch.

bblanchon, 2024-02-04:
I asked the ArduinoJson Assistant to write the program to generate the same content as the [cfg-tmp.json](https://github.com/bblanchon/ArduinoJson/files/14126698/cfg-tmp.json) you uploaded earlier.

Then, I ran my script again to see the effect of each `FORCE_INLINE` on the size of the code.
This time, I compiled for ESP8266, not AVR.

File:line                      | WLED
-------------------------------|-----
Array/ElementProxy.hpp:25      | 0
Array/ElementProxy.hpp:47      | 0
Array/JsonArray.hpp:65         | 448
Array/JsonArray.hpp:72         | 0
Array/JsonArray.hpp:78         | 0
Array/JsonArray.hpp:86         | 0
Array/JsonArray.hpp:92         | 0
Array/JsonArray.hpp:107        | 0
Array/JsonArray.hpp:113        | 0
Array/JsonArray.hpp:125        | 0
Array/JsonArray.hpp:135        | 0
Array/JsonArray.hpp:141        | 0
Array/JsonArray.hpp:147        | 0
Array/JsonArray.hpp:153        | 0
Document/JsonDocument.hpp:182  | 0
Document/JsonDocument.hpp:192  | 0
Document/JsonDocument.hpp:202  | 0
Document/JsonDocument.hpp:211  | 0
Document/JsonDocument.hpp:217  | 0
Document/JsonDocument.hpp:242  | 0
Document/JsonDocument.hpp:249  | 0
Document/JsonDocument.hpp:255  | 0
Document/JsonDocument.hpp:262  | 0
Document/JsonDocument.hpp:271  | 0
Document/JsonDocument.hpp:278  | 0
Document/JsonDocument.hpp:282  | 0
Json/Latch.hpp:28              | 0
Object/JsonObject.hpp:45       | 0
Object/JsonObject.hpp:51       | 0
Object/JsonObject.hpp:57       | 0
Object/JsonObject.hpp:63       | 0
Object/JsonObject.hpp:69       | 0
Object/JsonObject.hpp:77       | 0
Object/JsonObject.hpp:89       | 0
Object/JsonObject.hpp:105      | 0
Object/JsonObject.hpp:115      | 3520
Object/JsonObject.hpp:124      | 0
Object/JsonObject.hpp:131      | 0
Object/JsonObject.hpp:139      | 0
Object/JsonObject.hpp:147      | 0
Object/JsonObject.hpp:157      | 0
Object/MemberProxy.hpp:32      | 5952
Object/MemberProxy.hpp:38      | 1824
Object/MemberProxy.hpp:48      | 0
Object/MemberProxy.hpp:54      | -384
Strings/Adapters/RamString.:30 | -64
Variant/VariantAttorney.hpp:25 | 0
Variant/VariantAttorney.hpp:31 | -384
Variant/VariantRefBase.hpp:49  | 0
Variant/VariantRefBase.hpp:62  | 0
Variant/VariantRefBase.hpp:86  | 0
Variant/VariantRefBase.hpp:93  | 0
Variant/VariantRefBase.hpp:102 | 6032
Variant/VariantRefBase.hpp:107 | 1824
Variant/VariantRefBase.hpp:111 | 0
Variant/VariantRefBase.hpp:117 | 0
Variant/VariantRefBase.hpp:138 | 0
Variant/VariantRefBase.hpp:145 | 0
Variant/VariantRefBase.hpp:151 | 0
Variant/VariantRefBase.hpp:158 | 0
Variant/VariantRefBase.hpp:167 | 0
Variant/VariantRefBase.hpp:175 | 0
Variant/VariantRefBase.hpp:180 | 0
Variant/VariantRefBase.hpp:186 | 0
Variant/VariantRefBase.hpp:192 | 0
Variant/VariantRefBase.hpp:199 | 0
Variant/VariantRefBase.hpp:256 | 0
Variant/VariantRefBase.hpp:260 | 0
Variant/VariantRefBase.hpp:264 | -384
Variant/VariantRefBase.hpp:268 | 0
Variant/VariantRefBase.hpp:270 | 0
Variant/VariantRefBase.hpp:274 | 3344

I will soon edit this message to add columns for the ArduinoJson examples.

willmmiles, 2024-02-04:
> We can see that some of them have a very negative impact on code size.

Yes, I'm thinking there's a matter of scale here -- if we inline a function a few times, it doesn't matter much, but if we have to inline it hundreds of times, the extra code can really add up.  To make things even more complex, large projects like WLED use ArduinoJson in several different contexts (and more relevantly, translation units).  Being able to share common code can be a huge savings.

Is your script knocking out FORCE_INLINE or adding it only on the one line?  I'm running a data collection now over the whole WLED project, selectively removing FORCE_INLINE in each location and collecting program size and stack usage metrics.   I might try going the other way (selectively adding it) for comparison.

bblanchon, 2024-02-04:
My script removes one `FORCE_INLINE` at a time, so it compiles with `n-1` `FORCE_INLINE`s.
You might indeed get a different result if you compile with one `FORCE_INLINE` at a time.

Before making your tests, do you want me to push a new version without the seven `FORCE_INLINE`s that increase the code size?

willmmiles, 2024-02-04:
Data collection is done, I'm processing the results now.   I'll post them up shortly.  We can re-run with the changes after I've done the initial analysis.

willmmiles, 2024-02-04:
Apologies for the monster table.  This is the initial result set:

- The "flash" column shows the total ROM size difference with respect to the baseline (a3454e35)
- The other columns show changes to the stack size for each of the named functions.  Positive numbers increase stack size, negative numbers decrease it.  N/A is shown if the function has a stack size less than 128 bytes.
- The "no_inline" row globally disables FORCE_INLINE and lets the compiler decide.

| File:Line | Flash | wled00/hue.cpp:onHueData(void*, AsyncClient*, void*, size_t) | wled00/FX_fcn.cpp:WS2812FX::loadCustomPalettes() | wled00/json.cpp:deserializeSegment(ArduinoJson::V702PA2::JsonObject, byte, byte) | wled00/cfg.cpp:deserializeConfig(ArduinoJson::V702PA2::JsonObject, bool) | wled00/json.cpp:deserializeState(ArduinoJson::V702PA2::JsonObject, byte, byte) | wled00/json.cpp:serializePalettes(ArduinoJson::V702PA2::JsonObject, int) | wled00/cfg.cpp:_Z15serializeConfigv$part$0() | wled00/json.cpp:serializeModeNames(ArduinoJson::V702PA2::JsonArray) | wled00/json.cpp:serializeModeData(ArduinoJson::V702PA2::JsonArray) | wled00/cfg.cpp:_Z18serializeConfigSecv$part$0() | wled00/json.cpp:serializeInfo(ArduinoJson::V702PA2::JsonObject) | wled00/json.cpp:serveJson(AsyncWebServerRequest*) | wled00/set.cpp:handleSettingsSet(AsyncWebServerRequest*, byte) | wled00/ws.cpp:wsEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, uint8_t*, size_t) | wled00/ir.cpp:decodeIRJson(uint32_t) | wled00/json.cpp:serializeSegment(ArduinoJson::V702PA2::JsonObject&, Segment&, byte, bool, bool) | wled00/json.cpp:serializeState(ArduinoJson::V702PA2::JsonObject, bool, bool, bool, bool) | wled00/remote.cpp:remoteJson(int) |
| -----  | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- | ---------- |
| Array/ElementProxy.hpp:25 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/ElementProxy.hpp:47 | 272 | 16 | 48 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:107 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:113 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:125 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:135 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:141 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:147 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:153 | -32 | 0 | 16 | 32 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:65 | -1680 | 0 | 0 | 0 | 0 | 0 | 0 | -304 | 0 | 0 | 0 | -48 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:72 | -608 | 0 | 0 | 0 | 0 | 0 | -224 | 0 | -16 | -32 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:78 | -32 | 0 | 0 | 0 | -32 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:86 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Array/JsonArray.hpp:92 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:182 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:192 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:202 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:211 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:217 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:242 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:249 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:255 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:262 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:271 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:278 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Document/JsonDocument.hpp:282 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Json/Latch.hpp:28 | 176 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Object/JsonObject.hpp:105 | -112 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Object/JsonObject.hpp:115 | -2672 | 0 | 0 | 0 | 0 | 16 | 16 | -1312 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 16 | 0 |
| Object/JsonObject.hpp:124 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Object/JsonObject.hpp:131 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Object/JsonObject.hpp:139 | 64 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
[... cut ...]
| Variant/VariantRefBase.hpp:49 | -1456 | 16 | 32 | 0 | -288 | -48 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | -192 | 0 | N/A | 0 | 0 | N/A |
| Variant/VariantRefBase.hpp:62 | -432 | 16 | 0 | 16 | -256 | -48 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Variant/VariantRefBase.hpp:86 | 64 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| Variant/VariantRefBase.hpp:93 | -128 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| no_inline | -16992 | 0 | 64 | 0 | -224 | -16 | -240 | -3952 | -16 | -32 | -112 | -928 | -32 | -400 | 0 | N/A | N/A | N/A | N/A |

willmmiles, 2024-02-04:
Script code in context is here: https://github.com/willmmiles/WLED/blob/arduinjson-v7-dev/arduinojson_test.py

bblanchon, 2024-02-04:
Thank you for this fantastic work!  ❤️ 

I thought you wanted to compile with only one `FORCE_INLINE` at a time, but after looking at the source code, it seems you used the same technique as me: remove one `FORCE_INLINE` at a time.
Note that I subtracted the size from the baseline, so my results are reversed compared to yours.

I initially thought we could beat the compiler by adding judiciously placed `FORCE_INLINE`s, but I'm starting to believe this is a risky business.
On the one hand, `VariantRefBase.hpp:270` reduces the code by 352 bytes and saves 160 bytes for `deserializeSegment()`.
But on the other hand, `VariantRefBase.hpp:102` increases the code size by 13KB and loses `3824` bytes of stack in `serializeConfig()`.
In this game, there seems to be much more to lose than gain. 

After seeing your results, I think I'd better remove all the `FORCE_INLINE`, even if it means a 2% increase in the size of the examples.
Sure, I'll be unable to optimize the size of my benchmarks, but I'll avoid making unpredictable mistakes in everyone else's projects.

Let me know what you think.

willmmiles, 2024-02-04:
> I thought you wanted to compile with only one FORCE_INLINE at a time, but after looking at the source code, it seems you used the same technique as me: remove one FORCE_INLINE at a time.

Ah, sorry for the confusion - I wanted to try the knock-out approach, like you did, first; I was thinking about trying "start with none, add one at a time" as a follow up.   Given the test build framework, I had a half-baked idea to run a simulated annealing type search.  Although I fear a deeper search might lend itself to overfitting for the test sample.

It's definitely interesting how removing even one FORCE_INLINE on the "critical path" can make such a difference to the output on a large project.

I think there may not be a one-size-fits-all solution.  It could be the right answer is some level of configurability, as much as I hate the complexity of tunables.  For example, in WLED, we have many translation units doing JSON, so there's a big size advantage to sharing code.   A smaller project might have just one translation unit, where sharable functions just waste space.  Alas, I don't think link-time optimization is available for many of these embedded platforms. :(

Bottom line:  I think letting the compiler make the call is probably the best for unknown arbitrary use cases.  Given that you've already got the markup in the code, though, I think having a `ARDUINOJSON_SMALL_PROJECT` macro option or something similar to force micro-optimization might still be handy many use cases.

bblanchon, 2024-02-05:
I finally decided to keep twelve `FORCE_INLINE`s that have a positive impact on code size and no negative impact on stack consumption.

I force-pushed this commit because I made a mistake in the previous one, so you may need to do a hard reset.

Thank you again for your indispensable contribution!

closed 2024-02-05 by bblanchon

bblanchon, 2024-02-05:
The fix is available in [ArduinoJson 7.0.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.0.3).

willmmiles, 2024-02-05:
Thank you!  It was a pleasure to work with you. :)

commit cb0dc94d (2024-02-05): Remove most `FORCE_INLINE`s

I kept only the ones that had a positive impact on code size AND no negative impact on stack memory.

Fixes #2046