bblanchon/ArduinoJson #2048: I'm getting No memory error on deserialization

zekageri, 2024-02-06:
I'm using ArduinoJson v7 with ESP32 Wrover-E (8mb external ram)
Here is my log

```cpp
[HsH_Server] - Error: NoMemory. Size: 117
[HsH_Server] - Raw data: {"path":"/getFresh","method":"GET","query":{},"body":{},"id":"0fhw9bmm0eh6a7yzy4ep36"}
[HsH_Server] - Available heap: 104788 bytes
[HsH_Server] - Available external ram: 4023547 bytes
```

Here is the function

```cpp
hshServerClient.onFrame([this](const char* frame, size_t len){
    JsonObject doc;
    DeserializationError err =  deserializeJson(doc,frame);
    if(err){
        Serial.printf("[HsH_Server] - Error: %s. Size: %d\n", err.c_str(), len);
        Serial.printf("[HsH_Server] - Raw data: %s\n", frame);
        Serial.printf("[HsH_Server] - Available heap: %d bytes\n", ESP.getFreeHeap());
        Serial.printf("[HsH_Server] - Available external ram: %d bytes\n", ESP.getFreePsram());
        return;
    }
    handleRequest(doc);
});
```

The lambda is running in a separate task which has `16 * 1024` stack size and the char* frame is totally complete. Arduinojson refuses to deserialize. Even if i pass in the length.

This produces the same result

```cpp
hshServerClient.onFrame([this](const char* frame, size_t len){
    JsonObject doc;
    DeserializationError err =  deserializeJson(doc,frame,len); // Passing len
    if(err){
        Serial.printf("[HsH_Server] - Error: %s. Size: %d\n", err.c_str(), len);
        Serial.printf("[HsH_Server] - Raw data: %s\n", frame);
        Serial.printf("[HsH_Server] - Available heap: %d bytes\n", ESP.getFreeHeap());
        Serial.printf("[HsH_Server] - Available external ram: %d bytes\n", ESP.getFreePsram());
        return;
    }
    handleRequest(doc);
});
```

bblanchon, 2024-02-07:
Hi @zekageri,

You passed a null `JsonObject` to `deserializeJson()`.
You should pass a `JsonDocument` instead.

```diff
- JsonObject doc;
+ JsonDocument doc;
  DeserializationError err =  deserializeJson(doc,frame);
```

Best regards,
Benoit

zekageri, 2024-02-07:
Oh god. It was a silly mistake. Thank you very much and sorry for the issue :D

closed 2024-02-07 by zekageri