bblanchon/ArduinoJson #2049: Parser can not handle value bigger then 4,294,967,295

Jercyvz, 2024-02-08:
**Description**
Hi. 
I'm trying to parse next JSON sting below and want to extract "FrequencyHertz" value
"{\"APIVersion\":\"2\",\"Data\":{\"AbsoluteAngleOfArrivalRadians\":6.981317007977318,\"AngleOfArrivalErrorRadians\":0.0,\"AngleOfArrivalRadians\":0.0,\"CorrelationKey\":\"3336226785\",\"DetectionCount\":12,\"EpochTimeMilliSeconds\":1702299450911,\"FrequencyHertz\":57000000000,\"Name\":\"FrSky\",\"Protocol\":\"ACCESS\",\"RSSI\":-67}"

Everything works fine unti value is set bigger then 4294967295.
And this time parser returns "Ok" status.

Can you help with this? 

Thanks,
Eugene.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns deserialization
4. The program doesn't check the error
5. `deserializeJson()` returns `Ok`
6. Program fails to extract values from the `JsonDocument`
7. Program doesn't use `DeserializationOption::Filter`
8. Serialized document starts with a curly brace
9. Issue is not a confusion between array and object

**Environment**
* Microcontroller: Renesas RA4M1 (Arm® Cortex®-M4) 
* IDE: Visual Studio code with Platform IO

**Reproduction code**
```c++
 DynamicJsonDocument doc(packetSize);
    DeserializationError error = deserializeJson(doc, packetBuffer);
    Serial.print("Error=");
    Serial.println(error.c_str());
    if (error.code() != DeserializationError::Ok) {
        if (state.RelayCanBeTurnedOff())
        {
            ErrorState();
            state.currentState = CurrentState::Error;
        }
    }
    else {
        unsigned long freqValue = doc["Data"]["FrequencyHertz"];
        int rssiValue = doc["Data"]["RSSI"];

        SetRelayLedState(rssiValue, freqValue);

        Serial.print("FreqValue=");
        Serial.println(freqValue);

        Serial.print("RSSI=");
        Serial.println(rssiValue);
    }
```

**Remarks**
I'm using Arduino UNO R4 minima

bblanchon, 2024-02-09:
Hi @Jercyvz,

`unsigned long` cannot store values above 4,294,967,295.
You must switch to `unsigned long long`.

This is a duplicate of #2024.

Best regards.
Benoit

Jercyvz, 2024-02-09:
OMG!!! That was silly question. Sorry for that. Thanks a lot for your fast answer. 
My problem resolved.

closed 2024-02-09 by Jercyvz

bblanchon, 2024-02-09:
You're welcome, @Jercyvz.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: