bblanchon/ArduinoJson #2050: JsonDocument is fine the first time, but empty the second time

JesperThorup, 2024-02-10:
**Describe the issue**  
I'm recieving an json message using MQTT. It's working fine, the first time.

what happens is, I send at payload using MQTT to my ESP32. The ESP then runs the newSchedule function, at deserializes the json object.

At the end og newSchedule, it calls the updateSchedule, and print my two start times.

Now, every 15 minutes, the updateSchedule get's called directly from my main loop, but now states the docScheduele["Items"] is empty (not printing anything.

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP32_wroom]

**Reproduction**  
```
StaticJsonDocument<1024> docScheduele;  

void newSchedule(String json)
{
  DeserializationError error = deserializeJson(docScheduele, json);

  // Test if parsing succeeds.
  if (error)
  {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }

  Serial.println("[Schedule]: updated.");

  updateSchedule();
}

void updateSchedule()
{

  for (JsonObject MY_item : docScheduele["Items"].as<JsonArray>())
  {
    String _startTime = MY_item["StartTime"];

    Serial.println("[Schedule]: " + _startTime);
  }
}

```

and the json body:
```
{
    "FUNCTION": "SCH",
        "Items": [
            {                
                "StartTime": "2024-02-09T03:00:00Z"                
            },
            {
                "StartTime": "2024-02-09T06:00:00Z"
            }
        ]
}
```

Can anyone point me to the problem?

bblanchon, 2024-02-11:
Hi @JesperThorup,

I'm sorry, but I don't see anything in the code snippet that could explain your issue.
The problem probably lies somewhere else.
Please try to provide an [MCVE](https://stackoverflow.com/help/mcve).

Best regards,
Benoit

JesperThorup, 2024-02-11:
Found out it was a memory issue.. Using a string and the data went missing, but using std::string, the date stayed..

closed 2024-02-11 by JesperThorup

bblanchon, 2024-02-12:
You're welcome, @JesperThorup 
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: