bblanchon/ArduinoJson #2051: ArduinoJson 7 assumptions about memory in modern processors

judge2005, 2024-02-11:
I have been carefully optimizing memory in my ESP32 program because of ArduinoJson 6 not being able to allocate 6144 bytes to parse a document and then I read the ArduinoJson 7 introduction which was telling me how much memory 32 bit processors have, but this has not been my experience with the ESP32, so I am a little concerned with the assumptions outlined for v7.

Let me explain my deep dive in to heap memory on the ESP32:

Firstly many people aren't aware that although there is an advertised ram size of 540K, some of that RAM is taken by program allocations (or quite a lot in my case) and permanently allocated stacks for however many RTOS tasks are running, and the remainder is split up into multiple different heaps. The heap in IRAM has to be accessed on 4 byte boundaries, so this is essentially unavailable to most application code. Another heap is preferentially used for DMA access. That leaves essentially one heap for general use.

When most people look at free memory in the ESP32 they use the ESP.getFreeHeap() function. This returns the sum of all free memory in all the heaps. For example in my application it returns about 100k, so you can see why I have been confused about why a document requiring a mere 6144 bytes couldn't get enough memory. In reality, in my application 43k of that heap is in IRAM, so is unavailable to ArduinoJson. That still leaves 57k! Well yes, until I run WiFiClientSecure to open the connection to the server I need to retrieve the JSON document from. That takes another 40k. So that leave me with 17k. Suddenly, that's not looking like much RAM at all, and when I print out the maximum size contiguous block, that is about 8k. That's cutting it fine for my 6k document.

Then as my application continues to be used, it slowly permanently allocates about another 4k. Now this could be made non-permanent, but then I would be open to more heap fragmentation, which I have spent a good period of time slowly reducing, where I have control of the code. I could also shave some bytes of the stack size of my tasks, but hopefully you get the point that I am spending a lot of my time optimizing memory usage to try and ensure that ArduinoJson will have enough after WiFiClientSecure has gobbled up almost all of the available 8-bit aligned memory.

Anyway, now I have 13k at the point that I need to allocate 6k for my document deserializer and about 50% of the time, there is not a large enough contiguous block.

So I am spending a lot of time trying to optimize memory usage on this 32 bit device, and this is why I am a little alarmed when you say that modern processors have a lot of RAM. Hopefully I'm wrong to be alarmed, and you will tell me that v7 will actually improve matters for me, for example I see you have some optimizations in there such as de-duping strings that will certainly help in my case as the document I am parsing is largely made of multiple instances of the same object.

BTW, if you could use the memory in IRAM, that would be wonderful, but would obviously require some interesting coding to make sure all the memory accesses where aligned to 4 bytes!

Apologies if creating an issue is the wrong way to open this conversation, but I couldn't see any other way, and thanks for reading!

bblanchon, 2024-02-12:
Hi @judge2005,

> I read the ArduinoJson 7 introduction which was telling me how much memory 32 bit processors have

I'm assuming you're talking about [this article](https://arduinojson.org/news/2024/01/03/arduinojson-7/), where I say "32-bit microcontrollers have much more memory".
In this sentence, as in the whole "code size" section, I was talking about the Flash memory, not the RAM.
Indeed, the new UNO has eight times more Flash than the previous generation.

I was talking about the Flash memory, but even if I were talking about RAM, it would still be true: the new UNO has sixteen times more RAM than the previous generation.
But I get your point: the board has more RAM, but a huge part is already consumed or reserved.

Fortunately, ArduinoJson 7 doesn't consume more RAM than ArduinoJson 6, as you can see with the [ArduinoJson Assistant](https://arduinojson.org/v7/assistant/).
Moreover, ArduinoJson 7 doesn't need a contiguous memory block, so it can run when the heap is fragmented.

Please give it a try, and let me know if you have any problems.

Best regards,
Benoit

judge2005, 2024-02-13:
Thanks. I tried AduinoJson 7 and it is a definite improvement. I didn't notice any increase in firmware size and it definitely handles low memory (well, actually fragmented memory) better. I forced my application to do all the memory-hungry things and it never went wrong. At the time that it creates the 6k JSON document (or what previously required 6k) it worked fine with this `Free heap: 12368, Max block size=3060`, so a definite improvement.

Many thanks.

bblanchon, 2024-02-13:
I'm very glad to read that. Thanks!
I think I can safely close this issue then.
Don't hesitate to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2024-02-13 by bblanchon