bblanchon/ArduinoJson #2052: JsonArray.size contains always 0 after successful deserialization

chess-levin, 2024-02-11:
I have a JSON array as input. The file is stored in SPIFFS as a favorite list . Here is a shortened version of my `stations.json` file:

```
[
  {
    "name": "Cork 96fm",
    "url": "https://wg.cdn.tibus.net/96fm"
  },
  {
    "name": "TopBlues (US)",
    "url": "http://us3.internet-radio.com:8342/stream"
  },
  {
    "name": "McSlons Irish Pub Radio",
    "url": "http://server9.streamserver24.com/stream/macslonsradio/stream"
  }
]
```

My goal is to deserialize the file to an array of this struct

```
struct StationInfo {
  const char* name;
  const char* url;
};

StationInfo favArr[];
``` 

Deserialization works successful (see code an sample output). But I can't figure out how to get the number of array elements the JsonDocument holds after deserializing from json-file. I need the number of elements to dynamically allocate memory for my `favArr[]`.

```
  JsonArray array = doc.to<JsonArray>();
  Serial.print("Doc contains count items: "); Serial.println(array.size());
```

This code always returns 0.

Do you have any tip how to get the number of elements and how to dynamically allocate the correct chunk of memory?

Many thanks in advance.

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32 Dev Kit C V4
* Core/runtime: Arduino v3.0.2
* IDE: PlatformIO

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
   
void loadStations(const char *filename) {
  Serial.printf("Reading stations from: %s\n", filename); 
  
  File file = SPIFFS.open(filename);

  if(!file) {
    Serial.printf("ERROR: Failed to open: %s\n", filename);
    return;
  } 

  Serial.printf("Successfully opened: %s (%d bytes)\n", filename, file.size());
  
  JsonDocument doc;

  DeserializationError error = deserializeJson(doc, file);

  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }

  Serial.println("Successfully deserialized JSON");
  
  for (JsonObject item : doc.as<JsonArray>()) {
    const char* name = item["name"]; // "Radio Bremen 1", "Radio Bremen 2", "Radio Bremen 4", "DR P4 ...
    const char* url = item["url"]; // "http://icecast.radiobremen.de/rb/bremeneins/live/mp3/128/stream.mp3", ...
    Serial.printf("%s\t %s\n", name, url);
  }

  JsonArray array = doc.to<JsonArray>();
  Serial.print("Doc contains count items: "); Serial.println(array.size());

  int i = 0;
  for (JsonVariant v : array) {
      i++;
      const char* name = v["name"]; // "Radio Bremen 1", "Radio Bremen 2", "Radio Bremen 4", "DR P4 ...
      const char* url = v["url"];

      Serial.printf("%d\t %s\t %s\n", i, name, url);
  }

  file.close();
}
```

**Program output**  

Actual output:

```
Reading stations from: /stations.json
Successfully opened: /stations.json (2997 bytes)
Successfully deserialized JSON
Radio Bremen 1   http://icecast.radiobremen.de/rb/bremeneins/live/mp3/128/stream.mp3
Radio Bremen 2   http://icecast.radiobremen.de/rb/bremenzwei/live/mp3/128/stream.mp3
Radio Bremen 4   http://icecast.radiobremen.de/rb/bremenvier/live/mp3/128/stream.mp3
DR P4 (Bornholm)         http://live-icy.gss.dr.dk/A/A06L.mp3
NDR2 (NDS)       http://www.ndr.de/resources/metadaten/audio/m3u/ndr2.m3u
NDR Info (HH)    http://www.ndr.de/resources/metadaten/audio/m3u/ndrinfo_hh.m3u
WDR2 (OWL)       http://wdr-wdr2-ostwestfalenlippe.icecastssl.wdr.de/wdr/wdr2/ostwestfalenlippe/mp3/128/stream.mp3
WDR5     https://wdr-wdr5-live.icecastssl.wdr.de/wdr/wdr5/live/mp3/128/stream.mp3
Einslive         http://www.wdr.de/wdrlive/media/einslive.m3u
DLF      http://st01.sslstream.dlf.de/dlf/01/128/mp3/stream.mp3
DLF Kultur       http://st02.sslstream.dlf.de/dlf/02/128/mp3/stream.mp3
DLF Nova         http://st03.sslstream.dlf.de/dlf/03/128/mp3/stream.mp3
DLF DokDeb       https://st04.sslstream.dlf.de/dlf/04/128/mp3/stream.mp3
ffn      http://player.ffn.de/radioffn.mp3
8080s    http://regiocast.streamabc.net/regc-80s80smweb2517500-mp3-192-1672667
KCWR     http://kcrw.streamguys1.com/kcrw_192k_mp3_on_air
FluxFM Hamburg   http://streams.fluxfm.de/flux-hamburg/mp3-320/audio/
FluxFM Alternative       http://streams.fluxfm.de/alternative/mp3-320/streams.fluxfm.de/
FluxFM 80er      http://streams.fluxfm.de/80er/mp3-320/streams.fluxfm.de/
FluxFM 70er      http://streams.fluxfm.de/70er/mp3-320/streams.fluxfm.de/
FluxFM 60er      http://streams.fluxfm.de/60er/mp3-320/streams.fluxfm.de/
NRW Schlagerradio        http://rnrw.cast.addradio.de/rnrw-0182/deinschlager/low/stream.mp3
NRW Rockradio    http://rnrw.cast.addradio.de/rnrw-0182/deinrock/low/stream.mp3
NRW 90er         http://rnrw.cast.addradio.de/rnrw-0182/dein90er/low/stream.mp3
Cork 96fm        https://wg.cdn.tibus.net/96fm
TopBlues (US)    http://us3.internet-radio.com:8342/stream
McSlons Irish Pub Radio  http://server9.streamserver24.com/stream/macslonsradio/stream
Doc contains count items: 0
```

chess-levin, 2024-02-11:
Or does it make sense to use the `JsonDocument` object as a holder of my stations list?

So I would pass a reference to the object to my load/deserialize function, like this

``` c++
void loadStations(const char *filename, JsonDocument& doc);
```

chess-levin, 2024-02-11:
Hi @bblanchon!
Thanks for your comment. You've changed your Troubleshooter and I've tried to understand how this could help me, but I didn't get it. 

How does it answer my main question: Why does this code give 0 items?

``` c++
JsonArray array = doc.to<JsonArray>();
Serial.print("Doc contains count items: "); Serial.println(array.size());
```

chess-levin, 2024-02-11:
I'm now using these two global variables

``` c++
stationInfo_t * stationInfo;  // array of structures
int countElements = -1;     // array_len
``` 

and changed my function `int loadStations(const char *filename)` using `new` to allocate memory for my array of this struct

``` c++
typedef struct stationInfo_s {
  const char* name;
  const char* url;
} stationInfo_t;
``` 

``` c++
int loadStations(const char *filename) {
  Serial.printf("Reading stations from: %s\n", filename); 
  
  File file = SPIFFS.open(filename);

  if(!file) {
    Serial.printf("ERROR: Failed to open: %s\n", filename);
    return countElements;
  } 

  Serial.printf("Successfully opened: %s (%d bytes)\n", filename, file.size());
  
  JsonDocument doc;

  DeserializationError error = deserializeJson(doc, file);

  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return countElements;
  }

  JsonArray arr = doc.as<JsonArray>();
  countElements = arr.size();
  Serial.printf("Successfully deserialized %d elements\n", countElements);

  stationInfo = new stationInfo_t[countElements];

  int i = 0;
  for (JsonObject item : doc.as<JsonArray>()) {    
    stationInfo[i].name = item["name"];
    stationInfo[i].url = item["url"];
    Serial.printf("%d\t %s\t %s\n", i, stationInfo[i].name, stationInfo[i].url);
    i++;
  }

  file.close();
  return countElements;
}
``` 

Still got *one* question. Do I have to copy the strings from the `JsonDocument`  `item["name"]` and `item["url"]` or can I use the exisiting ones by using their adress?

chess-levin, 2024-02-12:
Answering my own question. Yes, I do have to duplicate the strings.

closed 2024-02-12 by chess-levin

bblanchon, 2024-02-12:
Hi @chess-levin,

I didn't find the time to tell you about `as<JsonArray>()`, but fortunately, you saw my commit for the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/).

About the configuration structure, I would avoid `strdup()` because it's a recipe for memory leaks.
Instead, I would use fixed-sized char arrays (as shown in [JsonConfigFile.ino](https://arduinojson.org/v7/example/config/)) or `String`s.

Best regards,
Benoit

chess-levin, 2024-02-12:
Thanks for your feedback. I've learned a lot yesterday ;-)
As I plan to read the presets just once (@ setup function) from the json-file and then hold them unchanged in memory, I think the risk of memory leaks is low.