bblanchon/ArduinoJson #2053: Get maximum JSON string length

danie1kr, 2024-02-12:
**Is your feature request related to a problem? Please describe.**  
Upgrading to v7 I discovered that some of my strings are not transported and report `null` instead.
Debugging a bit, I found `ARDUINOJSON_STRING_LENGTH_SIZE` defaulting to `2`, supporting up to 65k characters per string but my string had about 117k.

**Describe the solution you'd like**  
Add a simple function to get the maximum String length like:
```cpp
// feel free to adapt to your library style, just an idea
constexpr size_t maximumStringLength()
{
    return numeric_limits<uint_t<ARDUINOJSON_STRING_LENGTH_SIZE * 8>::type>::highest()
}
```
I could then chunk my string accordingly.

**Describe alternatives you've considered**  
Changing `ARDUINOJSON_STRING_LENGTH_SIZE` before including but this might not be the best option for others.

**Additional context**  
Thanks for the cool work!

bblanchon, 2024-02-13:
Hi @danie1kr,

Thank you very much for reporting this issue.

The undocumented constant `ArduinoJson::detail::StringNode::maxLength` already contains the maximum length of strings.
I don't plan on adding a public-facing function for this because I doubt anyone else would use it.

I added the relevant information to the [known issues](https://arduinojson.org/v7/known-issues/) and to the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/).

Best regards,
Benoit

closed 2024-02-13 by bblanchon

danie1kr, 2024-02-13:
Thanks for pointing out `ArduinoJson::detail::StringNode::maxLength`, I did not look in the `detail` namespace :|