bblanchon/ArduinoJson #2054: Thingsboard serialize json data issue

gittimotap, 2024-02-12:
**Describe the bug**  
https://github.com/thingsboard/thingsboard/issues/10164

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v7/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I used:
* Microcontroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
JsonDocument doc;

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

[insert repro code here]
```

**Compiler output**    
If relevant, include the complete compiler output (i.e. not just the line that contains the error.)

**Program output**  
If relevant, include the repro program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

gittimotap, 2024-02-12:
6.21.5 works, seems like the next release broke something. Found originally here: https://github.com/thingsboard/thingsboard/issues/10164

bblanchon, 2024-02-14:
Hi @gittimotap,

I need more information.
Please provide at least a reproduction code.

Best regards,
Benoit

gittimotap, 2024-02-14:
Hi Benoit
Easiest way is to take for example this https://github.com/thingsboard/thingsboard-client-sdk/blob/master/examples/0003-esp8266_esp32_send_data/0003-esp8266_esp32_send_data.ino and run it on some Arduino board (I tried both https://wiki.dfrobot.com/SKU_DFR0868_Beetle_ESP32_C3 and https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654), it will give you an error straight away. You can use demo.thingsboard.io, you can have a free account there.

In any case, I tried multiple of the SDK examples and all give the same error: https://github.com/thingsboard/thingsboard-client-sdk/tree/master/examples (ESP32 based, not ESP8266). I always get "[TB] Unable to serialize json data" with anything above 6.21.5.

Regards
Timo

commit f17fc055 (2024-02-18): Make `JSON_STRING_SIZE(N)` return `N+1` to fix third-party code

ThingsBoard uses this macro to compute size of char arrays ಠ_ಠ
https://github.com/thingsboard/thingsboard-client-sdk/blob/v0.12.2/src/Helper.h#L38

Closes #2054
closed 2024-02-18 by bblanchon