bblanchon/ArduinoJson #2055: Unsopported fields in `library.properties`

martin-eden, 2024-02-16:
Hello Benoit and thank you for high level of this library.

I'm writing Lua parser/compiler of `library.properites` and noticed that [your file](https://github.com/bblanchon/ArduinoJson/blob/7.x/library.properties#L10) has fields `repository` and `license` not covered by [official specification](https://arduino.github.io/arduino-cli/0.35/library-specification/#15-library-format-rev-22).

`repository` and `license` are present in `~/.arduino15/library_index.json`. I didn't find info on that format and think that it's ArduinoIDE internal kitchen.

-- Martin

bblanchon, 2024-02-17:
Hi @martin-eden,

These fields are for Particle.

See:

* [library.properties fields](https://docs.particle.io/getting-started/device-os/firmware-libraries/#library-properties-fields) on docs.particle.io
* #662

Best regards,
Benoit

martin-eden, 2024-02-17:
Thanks for clarification, Benoît. I'm closing issue.

closed 2024-02-17 by martin-eden