bblanchon/ArduinoJson #2064: switching  from DynamicJsonDocument to JsonDocument

MHz000, 2024-03-02:
I use an esp32doit-devkit-v1 with the bblanchon/ArduinoJson@^7.0.3 library. If it helps, I'll be happy to send the source code too
When I switch from DynamicJsonDocument to JsonDocument I get the following error messages:

src\main.cpp:147:24: required from here
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:40:3: error: no type named 'type' in 'struct ArduinoJson::V703PB2::detail::enable_if<false ,void>'
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note: candidate: ArduinoJson::V703PB2::JsonDocument::JsonDocument(ArduinoJson::V703PB2::JsonDocument&&)
    JsonDocument(JsonDocument&& src)
    ^
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note: no known conversion for argument 1 from 'int' to 'ArduinoJson::V703PB2::JsonDocument&&'
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note: candidate: ArduinoJson::V703PB2::JsonDocument::JsonDocument(const ArduinoJson::V703PB2::JsonDocument& )
    JsonDocument(const JsonDocument& src) : JsonDocument(src.allocator()) {
    ^
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note: no known conversion for argument 1 from 'int' to 'const ArduinoJson::V703PB2::JsonDocument&'
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:24:12: note: candidate: ArduinoJson::V703PB2::JsonDocument::JsonDocument(ArduinoJson::V703PB2::Allocator* ) <near match>
    explicit JsonDocument(Allocator* alloc = detail::DefaultAllocator::instance())
             ^
.pio\libdeps\esp32doit-devkit-v1\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:24:12: note: conversion of argument 1 would be ill-formed:
src\main.cpp:147:24: error: invalid conversion from 'int' to 'ArduinoJson::V703PB2::Allocator*' [-fpermissive]
    JsonDocument doc(1536); // Dynamic JSON?
                         ^
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
[image]
[image]

bblanchon, 2024-03-03:
Hi @MHz000,

When you get an error when compiling a C++ program, you must look at the first error in the compiler output (which might not be the order shown in the IDE).
In your case, the first error is:

```
error: no matching function for call to 'ArduinoJson::V703PB2::JsonDocument::JsonDocument(int)'
```

This error tells that `JsonDocument` has no constructor with an `int` parameter.
You can fix it like so:

```diff
- JsonDocument doc(1536);
+ JsonDocument doc;
```

If you think a document was unclear and led you to make this error, please let me know so I can fix it immediately.

Best regards,
Benoit

MHz000, 2024-03-03:
BINGO!
Hi Benoit, your tip solved my problem straight away, thank you!;
You will have seen at first glance that I am not an experienced C++ programmer. So I dare to ask you another question. In my program I receive an extensive JSON file see appendix, which I use to decode by
<code>
if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
                         String payload = https.getString();
                         Serial.println("GOT API RESPONSE");
                        decodeJsonPayload(payload.c_str());
</code>
take apart
<code>
void decodeJsonPayload(const char* payload) {
   JsonDocument doc;
   DeserializationError error = deserializeJson(doc, payload);
   if (error) { ...
</code)
My problem: I only get the last part of the deserialization_ed payload as a result.
If it helps with the solution I would be happy to send the entire source code#
Thank you
Joschen
[JSON-data.txt](https://github.com/bblanchon/ArduinoJson/files/14473093/JSON-data.txt)

bblanchon, 2024-03-05:
What does the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/) say?

MHz000, 2024-03-05:
Hi Beniot
thanks to your hint
"You can fix it like this:
- JsonDocument doc(1536);
+ JsonDocument doc;"
I can now use the JSON document without any problems. You can find a JSON example in the appendix. My question boils down to this: How can I access the individual sub-documents {timestamp...}, {timestamp...},... separately? Yes, it is possible to use String operators to do this. This is tedious, error-prone and boring. Is there a more elegant way.
Thank you for your great library and support
Joschen
[JSON-data.txt](https://github.com/bblanchon/ArduinoJson/files/14496060/JSON-data.txt)

bblanchon, 2024-03-07:
Hi Joschen,

You can find extensive documentation at https://arduinojson.org/
Let me know if you have any issues.

Best regards,
Benoit