bblanchon/ArduinoJson #2067: Is there a way to cache a JsonObject between function calls?

scottchiefbaker, 2024-03-13:
I have Googled, looked in closed issues, and I even hit up ChatGPT and I cannot find an answer to what I assume is a simple request.

**Describe the issue**  
I need a way to cache a JsonObject between function calls to reuse if a request is made too quickly

**Environment**  
Here is the environment that I'm using':
* Microconroller: Wemos D1 Mini
* Core/runtime: ESP8266 core for Arduino v3.1.2
* IDE: `arduino-cli`

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
#include "ArduinoJson.h"

void foobar(JsonObject obj) {
    static JsonObject last_obj;
    static uint32_t last_hit = 0;
    bool too_soon            = (millis() - last_hit) < 5000;

    if (too_soon) {
        // Use the last cached hit somehow?
        obj.set(last_obj);
    } else {
        obj["name"]  = "Jason Doolis";
        obj["color"] = "Red";

        last_hit = millis();
        last_obj.set(obj); // Cache the obj for next time?
    }
}

void setup() {
    Serial.begin(115200);
}

void loop() {
    JsonDocument json;

    JsonObject xx = json["test"].to<JsonObject>();
    foobar(xx);

    serializeJson(json, Serial);
    Serial.println();

    delay(2000);
}
```

**Expected output:**
The JSON uses the previous iteration as a cache

**Actual output:**
When I call this function every two seconds I see this:

```
{"test":{"name":"Jason Doolis","color":"Red"}}
{"test":{}}
{"test":{}}
{"test":{"name":"Jason Doolis","color":"Red"}}
{"test":{}}
{"test":{}}
```

Clearly it's getting the correct data the first time, but when it comes back and triggers the `too_soon` it's not seeing the data from the previous iteration in `last_obj`. Bonus points if I could append `"cached": true` to the object after realize we're in the `too_soon` if statement.

bblanchon, 2024-03-15:
Hi @scottchiefbaker,

[`JsonObject`](https://arduinojson.org/v7/api/jsonobject/) is a reference to an object in a [`JsonDocument`](https://arduinojson.org/v7/api/jsondocument/).
In other words, `JsonObject` doesn't hold any data but points to the data in the `JsonDocument`.

To establish this link from the `JsonObject` to the `JsonDocument`, you must assign the reference to something that exists in the document; otherwise, the reference will be unbound.

In your case, `last_obj` is unbound because it is never assigned.
Sure, the program calls [`JsonObject::set()`](https://arduinojson.org/v7/api/jsonobject/set/), but it's a no-op because the reference is unbound.

One solution could be to assign `last_obj`, like so:

```diff
- last_obj.set(obj);
+ last_obj = obj;
```

But I suspect that caching the reference is not what you are after.
Instead, you probably expect to save the *content* of the object pointed by `obj`.
In that case, the solution is to use a `JsonDocument` instead:

```diff
- static JsonObject last_obj;
+ static JsonDocument last_obj;
```

Best regards,
Benoit

bblanchon, 2024-03-16:
Indeed, [`JsonObject::set()`](https://arduinojson.org/v7/api/jsonobject/set/) only accepts `JsonObjectConst`, and there is no implicit conversion between `JsonDocument` and `JsonObjectConst`. I should probably improve that, but I never noticed it because very few people use `JsonObject::set()`.

You can workaround this error by explicitly converting the `JsonDocument` to a `JsonObject`:

```diff
- obj.set(last_obj);
+ obj.set(last_obj.as<JsonObject>());
```

Alternatively, you could refactor your program so that `foobar()` returns a `JsonDocument`:

```c++
JsonDocument foobar() {
    static JsonDocument doc;
    static uint32_t last_hit = 0;
    bool too_soon = (millis() - last_hit) < 5000;

    if (!too_soon) {
        doc["name"]  = "Jason Doolis";
        doc["color"] = "Red";
        last_hit = millis();
    }

    return last_obj;
}

void loop() {
    JsonDocument json;
    json["test"] = foobar();

    serializeJson(json, Serial);
    Serial.println();

    delay(2000);
}
```

Sure, it produces one or two additional copies, but it's a thousand times more readable.

Another alternative would be to return a `String` from `foobar` and then use [`serialized()`](https://arduinojson.org/v7/api/misc/serialized/) to insert it in the main document.

scottchiefbaker, 2024-03-16:
Aha! This is exactly what I needed. I didn't realize you could put a `JsonDocument` _into_ another `JsonDocument`. That **really** simplifies things. With this method I was able to include `"cached": true` as well. 

There is a minor typo in your code above with `last_obj` so here is my full working code for posterity in case someone else has a similar problem:

```c++
#include "ArduinoJson.h"

void setup() {
  Serial.begin(115200);
}

JsonDocument foobar() {
  static JsonDocument doc;
    
  static uint32_t last_hit = 0;
  int32_t diff             = millis() - last_hit;
  bool too_soon            = abs(diff) < 5000;

  if (last_hit && too_soon) {
    doc["cached"] = true;
  } else {
    doc.clear();
    doc["name"] = "Jason Doolis";
    doc["time"] = millis();

    last_hit = millis();
  }

  return doc;
}

void loop() {
  JsonDocument json;
  json["test"] = foobar();

  serializeJson(json, Serial);
  Serial.println();

  delay(2000);
}
```

Thank you for your assistance on this. And thank you for an AMAZING library. ArduinoJson is a masterpiece.

closed 2024-03-16 by scottchiefbaker

bblanchon, 2024-03-17:
You're welcome, @scottchiefbaker!
Thank you for using ArduinoJson.