bblanchon/ArduinoJson #2068: compilation error

mpsampaio, 2024-03-14:
**Description**
error occurs in compilation.
from C:\Users\mpsam\PROJETOS-ARDUINO\esp8266-fastled-webserver-main\esp8266-fastled-webserver\esp8266-fastled-webserver\esp8266-fastled-webserver.ino:19:
c:\Users\mpsam\OneDrive\Documentos\Arduino\projetos\libraries\ArduinoJson\src/ArduinoJson/Variant/Converter.hpp:9:1: error: pasting ")" and "2" does not give a valid preprocessing token
    9 | ARDUINOJSON_BEGIN_PUBLIC_NAMESPACE

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. The error is not in the list

**Environment**
* Microcontroller: ESP8266
* IDE: Arduino IDE 2.3.2

scottchiefbaker, 2024-03-14:
@mpsampaio this bug report isn't very helpful unless you post your code.

bblanchon, 2024-03-15:
In all likelihood, there is a problem with one of the following macros:

* [`ARDUINOJSON_DECODE_UNICODE`](https://arduinojson.org/v7/config/decode_unicode/)
* [`ARDUINOJSON_ENABLE_COMMENTS`](https://arduinojson.org/v7/config/enable_comments/)
* [`ARDUINOJSON_ENABLE_INFINITY`](https://arduinojson.org/v7/config/enable_infinity/)
* [`ARDUINOJSON_ENABLE_NAN`](https://arduinojson.org/v7/config/enable_nan/)
* [`ARDUINOJSON_ENABLE_PROGMEM`](https://arduinojson.org/v7/config/enable_progmem/)
* [`ARDUINOJSON_USE_DOUBLE`](https://arduinojson.org/v7/config/use_double/)
* [`ARDUINOJSON_USE_LONG_LONG`](https://arduinojson.org/v7/config/use_long_long/)

They can only be set to `0` or `1` and **nothing else**.

mpsampaio, 2024-03-15:
Thank you for your attention, sorry for the inconvenience. I switched to vscode/platformio and reviewed the dependencies. now the project compiles without errors.
thanks

closed 2024-03-15 by mpsampaio