bblanchon/ArduinoJson #2074: best way to create a sub document

briandilley, 2024-03-29:
Is there a better way of doing this?

```cpp
    JsonDocument doc;

    doc["config"]["tmp"] = true; 
    doc["config"].remove("tmp");
    JsonObject configCtxDoc = doc["config"].as<JsonObject>(); 

    // this method adds stuff to the configCtxDoc
    this->configCtx.getConfiguration(configCtxDoc);
```

The resulting object should look something like this:

```json
{
...
  "config": {
     ... stuff added by this->configCtx.getConfiguration(configCtxDoc)...
  }
...
}
```

I was hoping there'd be a way to create the sub document like so:
```cpp
  JsonDocument doc;
  this->configCtx.getConfiguration(doc["config"]);
```

or:
```cpp
  JsonDocument doc;
  this->configCtx.getConfiguration(doc.add<JsonObject>("config"));
```

bblanchon, 2024-03-30:
Hello @briandilley,

You must use [`to<JsonObject>()`](https://arduinojson.org/v7/api/jsonvariant/to/), like so:

```c++
JsonDocument doc;
this->configCtx.getConfiguration(doc["config"].to<JsonObject>());
```

Best regards,
Benoit

briandilley, 2024-03-30:
@bblanchon that doesn't work:

```cpp
    JsonDocument doc;
    doc["id"] = this->getId();
    doc["name"] = this->getName();
    doc["weight"] = this->getWeight();
    doc["demoable"] = this->isDemoable();

    this->configCtx.getConfiguration(doc["config"].as<JsonObject>());
```

Result:

```
src/LED_ShowManager.cpp:157:66: error: cannot bind non-const lvalue reference of type 'ArduinoJson::V703PB2::JsonObject&' to an rvalue of type 'ArduinoJson::V703PB2::detail::enable_if<true, ArduinoJson::V703PB2::JsonObject>::type' {aka 'ArduinoJson::V703PB2::JsonObject'}
```

bblanchon, 2024-04-02:
> cannot bind non-const lvalue reference of type `JsonObject&` to an rvalue of type `JsonObject`

Somewhere in your code, probably in the definition of `getConfiguration()`, there is a `JsonObject&` that must be replaced with `JsonObject`.

briandilley, 2024-04-03:
Ok - why should it always be passed by value? Does that not create a copy? The goal here was to pass an object to this function to have the function alter it.

briandilley, 2024-04-03:
[This is the answer I was looking for](https://arduinojson.org/v7/api/jsonobject/)

briandilley, 2024-04-03:
Closing this

closed 2024-04-03 by briandilley