bblanchon/ArduinoJson #2082: Not work

techn0man1ac, 2024-04-23:
**Description**
Press compile and return error

**Troubleshooter's report**
1. The program uses ArduinoJson 5
2. The issue happens at compile time
3. The error is not in the list

**Environment**
* Microcontroller: Arduino Nanp
* Core/Framework: Arduino 1.8.19
* IDE: Arduino 1.8.19

**Reproduction code**
```c++

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\razni\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\razni\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\razni\OneDrive\Документи\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\razni\AppData\Local\Temp\arduino_build_827954 -warnings=none -build-cache C:\Users\razni\AppData\Local\Temp\arduino_cache_191917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\razni\OneDrive\Документи\Arduino\libraries\ArduinoJson\examples\JsonHttpClient\JsonHttpClient.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\razni\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\razni\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\razni\OneDrive\Документи\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10819 -build-path C:\Users\razni\AppData\Local\Temp\arduino_build_827954 -warnings=none -build-cache C:\Users\razni\AppData\Local\Temp\arduino_cache_191917 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\razni\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\razni\OneDrive\Документи\Arduino\libraries\ArduinoJson\examples\JsonHttpClient\JsonHttpClient.ino
Using board 'nano' from platform in folder: C:\Users\razni\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\razni\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Detecting libraries used...
"C:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "C:\\Users\\razni\\AppData\\Local\\Temp\\arduino_build_827954\\sketch\\JsonHttpClient.ino.cpp" -o nul
Alternatives for ArduinoJson.h: [ArduinoJson@5.13.5]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@5.13.5]
"C:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\razni\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "-IC:\\Users\\razni\\OneDrive\\Документи\\Arduino\\libraries\\ArduinoJson\\src" "C:\\Users\\razni\\AppData\\Local\\Temp\\arduino_build_827954\\sketch\\JsonHttpClient.ino.cpp" -o nul
Alternatives for ArduinoJson.h: [ArduinoJson@5.13.5]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@5.13.5]
JsonHttpClient:19:10: fatal error: ArduinoJson.h: No such file or directory
compilation terminated.
Використовується бібліотека ArduinoJson версії 5.13.5 з теки: C:\Users\razni\OneDrive\Документи\Arduino\libraries\ArduinoJson 
exit status 1
ArduinoJson.h: No such file or directory

```

bblanchon, 2024-04-25:
Hi @techn0man1ac,

I just tried with a fresh install, and it works as expected.
Then I changed my sketchbook folder to `E:\Документи` and started to get the same error.
I suspect that the toolchain doesn't fully support Unicode folder names.

Please change your Arduino sketchbook to a folder with only ANSI characters, such as `C:\Users\razni\OneDrive\Arduino`.

Best regards,
Benoit

techn0man1ac, 2024-04-25:
When I manually downloaded and extracted the library to a directory
`C:\Program Files (x86)\Arduino\libraries\ArduinoJson`
And deleted the old library - everything worked. 

Thank you

closed 2024-04-25 by techn0man1ac

bblanchon, 2024-04-26:
You're welcome, @techn0man1ac.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: