bblanchon/ArduinoJson #2090: v7 equivalent of validateJson

higaski, 2024-05-14:
The v6 documentation contains a small [`validateJson`](https://arduinojson.org/v6/how-to/validate-json/) snippet which I have always found immensely useful:
```cpp
bool validateJson(const char* input) {
  StaticJsonDocument<0> doc, filter;
  return deserializeJson(doc, input, DeserializationOption::Filter(filter)) == DeserializationError::Ok;
}
```

Now, the v7 documentation contains an equivalent snippet, but I've red that Static- and DynamicJsonDocument got merged into a single class which now always allocates in 1kB blocks. Does this mean that `validateJson` now allocates 2x1kB?
```cpp
bool validateJson(const char* input) {
  JsonDocument doc, filter;
  return deserializeJson(doc, input, DeserializationOption::Filter(filter)) == DeserializationError::Ok;
}
```

bblanchon, 2024-05-15:
Hi @higaski,

The current implementation allocates the memory pools lazily, so an empty `JsonDocument` doesn't require any heap allocation.

You can confirm that by replacing the allocator:
https://wandbox.org/permlink/p07bfdAXL9johHby

Best regards,
Benoit

higaski, 2024-05-15:
Thanks for clearing that up!

closed 2024-05-15 by higaski