bblanchon/ArduinoJson #2097: Bring back zero-copy mode in v7

LeedanJohnson, 2024-05-23:
**Is your feature request related to a problem? Please describe.**  
I regularly deserialize large documents with many strings. Historically in v6 I could leverage zero-copy mode to avoid duplicating the strings when deserializing the documents. That saves lots of memory with these large string-heavy documents. This is currently the only thing stopping me from moving from v6 to v7.

**Describe the solution you'd like**  
I'm hoping that the bulk of the v6 implementation of zero-copy mode is easily migrated to v7. If that's the case, I'd propose reinstating that functionality, perhaps using a different function signature like deserializeJsonZeroCopy(const char*) to avoid ambiguity.

**Describe alternatives you've considered**  
The easy alternative is simply to stay on v6, which is where I currently am. That's fine, but I'd really like to take advantage of new features and optimizations in v7. If there is another way to reduce memory use with large string-heavy documents I'm definitely interested.

Thanks for your consideration.

bblanchon, 2024-05-24:
Hi @LeedanJohnson,

After supporting ArduinoJson users for more than 10 years, I can tell you that the zero-copy mode was largely misunderstood and systematically misused.
I'm not bringing it back.

Best regards,
Benoit

closed 2024-05-24 by bblanchon

bblanchon, 2024-05-24:
PS: Did you compare the memory consumption between ArduinoJson 6 and 7 using the ArduinoJson Assistant?

LeedanJohnson, 2024-05-24:
Very well, I can respect that decision. If it comes to it there are other steps I can take to help improve the situation for my use-case.

An example document shows that v6 will use 3.5kB and v7 will use 8.5kB. V7 also shows a peak usage number of 14.5kB that I'm not sure how to interpret.

I did just realize however that v7's usage of discrete 1k blocks likely removes the largest limitation I've experienced with v6 requiring contiguous memory, and may help offset the additional memory usage incurred by dropping zero-copy mode. I'll do some testing, but this may make an update to v7 work for me after all. The ~2x increase in memory use is unfortunate, but may be manageable.

Thanks as always for your consideration and feedback, and all the work you do in maintaining these libraries. Cheers!

bblanchon, 2024-05-24:
With ArduinoJson 6, it is 3.5kB **plus the size of the in buffer**.
With ArduinoJson 7, you can discard the input buffer or not use any buffer if the data comes from a stream.

Peak memory usage includes the overallocated memory used by `deserializeJson()` but is released when the function exits.