bblanchon/ArduinoJson #2098: MQTT Publish destroying Json

Mallo321123, 2024-05-27:
I think this Is a MQTT problem, but https://github.com/knolleary/pubsubclient looks a bit dead.

My ESP board: ESP8266

My problem: Every time I Publish a MQTT Message, a seperate Json gets destroyed.

Here is my test code:

    for (uint8_t i = 0; i < length_d; i++) {    //Here the Json is still Intact
      Serial.print((char)payload_d[i]);
    }
    Serial.println("    4");

    client.publish("temp/display", mqtt_message, true);     //Publishing a different Json

    for (uint8_t i = 0; i < length_d; i++) {    //Here the Json is Broken
      Serial.print((char)payload_d[i]);
    }
    Serial.println("    5");

Here is my Serial Output:
`22:54:21:415 -> {"time":2000,"data":[{"name":"Leistung","value":50,"unit":"W"},{"name":"Durchfluss","value":520,"unit":"L"},{"name":"test","value":234,"unit":"J"}]}    4`

`22:54:21:429 -> y{"temp_display":22.36099243,"humidity_display":57.86027908}"},{"name":"Durchfluss","value":520,"unit":"L"},{"name":"test","value":234,"unit":"J"}]}    5`

at number 4 the Json is still intact, but as one gets sent, it just gets overwritten by the sent one (number 5).

bblanchon, 2024-05-29:
Hi @Mallo321123,

This question is not about ArduinoJson.
Please address your question to the author of PubSubClient or to StackOverflow.
Don't forget to include all the information; for example, what is `payload_d`?

Best regards,
Benoit

closed 2024-05-29 by bblanchon

Mallo321123, 2024-06-01:
payload_d is the varable, where a recieved MQTT message gets stored