bblanchon/ArduinoJson #2101: load an object into an absolute memory address, Cortex-M.

autoprog, 2024-06-05:
hi,
Im tested example on Pi Pico.
String input = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}"; run ok.

I need to load an object into an absolute memory address,

example pseude code:
#define PIN102  (*((volatile unsigned long *) 0x2000000))
PIN102 = 123;
PIN102 += 1024... ok

   JsonDocument * doc  (*((JsonDocument  *) 0x20020008))  <---- or other address.

   deserializeJson(doc, input);
   long time = doc[String("time")];
    time += 3000;
    doc[String("time")] = time;
    bla, bla, bla

I need to work with memory outside of the main program's RAM area.

thanks,
Carlos.

bblanchon, 2024-06-06:
Hi Carlos,

> I need to work with memory outside of the main program's RAM area.

It sounds like you want to write a [custom allocator](https://arduinojson.org/v7/api/jsondocument/#allocator).

Best regards,
Benoit

autoprog, 2024-06-06:
hi bblanchon,
Thank you for your quick help.

Im tested other solution, with Class,
MyClass::MyClass()
{.....
//------------------------------------
int MyClass::get_doc( int idx)
{
  int temp1 = doc[String("val1")];
          temp1 += 8;
           doc[String("val1")] = temp1;
  return   doc[String("val1")];
}
//-----------------------------------
String MyClass::get_str( int idx)
{
     String sensor = doc["sensor"];
           sensor += "@";
           doc["sensor"] = sensor;  
  return   doc["sensor"];
}

MyClass  *myInstancePtr;
int *ptr = (int *) 0x20020008; pointer in ram
//--------------------
myInstancePtr = new  (ptr) MyClass();

 Serial.print("GET2doc: "); Serial.println( myInstancePtr->get_doc(1)); ok
  Serial.print("GET2str: "); Serial.println( myInstancePtr->get_str(1)); ok
//--------------

With the class I would already be able to work, but I will try the best practice.
since you want to work with several json at the same time.

Now I'm trying your solution but it's giving errors, but the problem is with the compiler,
But if you have a code example I would appreciate it.

thanks,
Carlos.

bblanchon, 2024-06-10:
Hi Carlos,

The placement new won't work with ArduinoJson 7 because `JsonDocument` will allocate memory on the heap.
However, it would work with ArduinoJson 6's `StaticJsonDocument`.

Still, I recommend customizing the allocator if you want control over the memory allocation.

Best regards,
Benoit

autoprog, 2024-06-10:
hi bblanchon,
but it would be great to have an option to control addresses, 
as for those who are going to deal with many documents at the same time it would be very good.
with Malloc it works well, but without control of the memory blocks,
In my json test inside Class new with Malloc it works and using free()
 and resuming everything is perfect.

thanks,
Carlos.

closed 2024-06-12 by autoprog