bblanchon/ArduinoJson #2104: Lint error ovl_no_viable_subscript has occured

ShortArrow, 2024-06-13:
[image]

**Describe the issue**  
Lint error has occured. Compile process is no issue.

**Troubleshooter report**  

```text
No viable overloaded operator[] for type 'JsonDocument', clang(ovl_no_viable_subscript).
Type 'JsonObject' does not provide a subscript operator, clang(ovl_no_oper).
No matching function for call to 'serializeJson'
No matching function for call to 'serializeJson', clang(ovl_no_viable_function_in_call).
JsonSerializer.hpp(131, 8): Candidate function template not viable: no known conversion from 'JsonDocument' to 'JsonVariantConst' for 1st argument
JsonSerializer.hpp(138, 15): Candidate function not viable: requires 3 arguments, but 2 were provided
```

**Environment**  
Here is the environment that I'm using':
* Microconroller: RA4M1
* Core/runtime: arduino:renesas_uno:minima
* arduino-cli  Version: 0.35.3-arch Commit: 95cfd654fe0dd6b07e903d3132c0faceabfbe9e3

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
#include <ArduinoJson.hpp>
#include "Arduino.h"
#include "ArduinoJson/Array/JsonArray.hpp"
#include "ArduinoJson/Document/JsonDocument.hpp"
#include "ArduinoJson/Object/JsonObject.hpp"

void CanbusToJsonSerial(String label, int value) {
  JsonDocument doc;
  JsonObject data = doc["data"].to<JsonObject>();
  doc["topic"] = "CanbusA";
  data[label] = value;
  doc.shrinkToFit();
  serializeJson(doc, Serial);
  Serial.println();
}
```

`arduino-cli.yaml`

```yaml
board_manager:
  additional_urls: []
build_cache:
  compilations_before_purge: 10
  ttl: 720h0m0s
daemon:
  port: "COM5"
directories:
  data: Arduino
  downloads: Arduino/staging
  user: Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: true
fqbn: arduino:renesas_uno:minima

```

`.clangd`

```
CompileFlags:
  Add: [
    "-I./**",
    "-I./Arduino/data/packages/arduino/tools/arm-none-eabi-gcc/**",
    "-I./Arduino/packages/arduino/hardware/renesas_uno/1.2.0/variants/MINIMA/includes/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/variants/MINIMA/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/libraries/Arduino_CAN/src/**",
    "-I./Arduino/packages/arduino/hardware/renesas_uno/1.2.0/libraries/Arduino_FreeRTOS/src/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/libraries/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/tinyusb/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/deprecated/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/deprecated-avr-comp/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/api/**",
    "-I./Arduino/data/packages/arduino/hardware/renesas_uno/1.2.0/cores/arduino/**",
    "-I./Arduino/packages/arduino/hardware/renesas_uno/1.2.0/libraries/**",
    "-I./Arduino/data/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/**",
    "-I./Arduino/**",
  ]

```

**Program output**  

```JSON
[{
	"resource": "./target.cpp",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "ovl_no_viable_subscript",
	"severity": 8,
	"message": "No viable overloaded operator[] for type 'JsonDocument'",
	"source": "clang",
	"startLineNumber": 391,
	"startColumn": 24,
	"endLineNumber": 391,
	"endColumn": 25
},{
	"resource": "./target.cpp",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "ovl_no_viable_subscript",
	"severity": 8,
	"message": "No viable overloaded operator[] for type 'JsonDocument'",
	"source": "clang",
	"startLineNumber": 392,
	"startColumn": 6,
	"endLineNumber": 392,
	"endColumn": 7
},{
	"resource": "./target.cpp",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "ovl_no_oper",
	"severity": 8,
	"message": "Type 'JsonObject' does not provide a subscript operator",
	"source": "clang",
	"startLineNumber": 393,
	"startColumn": 7,
	"endLineNumber": 393,
	"endColumn": 8
},{
	"resource": "./target.cpp",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "ovl_no_viable_function_in_call",
	"severity": 8,
	"message": "No matching function for call to 'serializeJson'",
	"source": "clang",
	"startLineNumber": 395,
	"startColumn": 3,
	"endLineNumber": 395,
	"endColumn": 16,
	"relatedInformation": [
		{
			"startLineNumber": 131,
			"startColumn": 8,
			"endLineNumber": 131,
			"endColumn": 21,
			"message": "Candidate function template not viable: no known conversion from 'JsonDocument' to 'JsonVariantConst' for 1st argument",
			"resource": "./Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp"
		},
		{
			"startLineNumber": 138,
			"startColumn": 15,
			"endLineNumber": 138,
			"endColumn": 28,
			"message": "Candidate function not viable: requires 3 arguments, but 2 were provided",
			"resource": "./Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp"
		}
	]
}]
```

bblanchon, 2024-06-14:
Hi @ShortArrow,

This is a problem with the linter, not with the library.
What do you expect from me?

Best regards,
Benoit

ShortArrow, 2024-06-14:
@bblanchon Thank you. I realized the problem was that the linter failed to detect operator overloading. Just being able to realize that was a huge win for me.

I'm not very good at cpp, so I couldn't read deep into the library. So big thanks for you.

closed 2024-06-14 by ShortArrow