bblanchon/ArduinoJson #2107: Question - Access JSON data variables in nested arrays

justinpanta, 2024-06-30:
Firstly, I am a beginner not only to ArduinoJSON, but also to C++. ArduinoJSON is an amazing library, however, due to my lack of knowledge in this area, I have a question that I'm hoping will be easily answered. 

I'm using ArduinoJSON on an ESP32 with Arduino IDE 1.8.19 to deserialize weather data from a JSON API. Using the ArduinoJSON assistant, it has generated the program code to use in my project, which works well.
I can successfully access and use any deserialized data that is assigned to variables or pointers (int, float, const char*), however, as some of the JSON source data includes items that are in arrays, I am having trouble working out how to access this data. As shown below, the data in particular that I am having trouble with is the 'forecasts', as it has forecast data for multiple days. I need to know how to select a particular day (1-3) and access the associated data for that day as variables.

Here is a sample of the JSON data I receive from the weather source:

{
    "location": {
        "id": 13973,
        "name": "Arthurs Creek",
        "region": "Yarra Valley and Dandenong Ranges",
        "state": "VIC",
        "postcode": "3099",
        "timeZone": "Australia/Melbourne",
        "lat": -37.579,
        "lng": 145.2014,
        "typeId": 21
    },
    "forecasts": {
        "weather": {
            "days": [
                {
                    "dateTime": "2024-06-30 00:00:00",
                    "entries": [
                        {
                            "dateTime": "2024-06-30 00:00:00",
                            "precisCode": "chance-shower-cloud",
                            "precis": "Shower or two",
                            "precisOverlayCode": "",
                            "night": false,
                            "min": 5,
                            "max": 10
                        }
                    ]
                },
                {
                    "dateTime": "2024-07-01 00:00:00",
                    "entries": [
                        {
                            "dateTime": "2024-07-01 00:00:00",
                            "precisCode": "partly-cloudy",
                            "precis": "Partly cloudy",
                            "precisOverlayCode": "",
                            "night": false,
                            "min": 3,
                            "max": 11
                        }
                    ]
                },
                {
                    "dateTime": "2024-07-02 00:00:00",
                    "entries": [
                        {
                            "dateTime": "2024-07-02 00:00:00",
                            "precisCode": "mostly-fine",
                            "precis": "Morning frost. Mostly sunny",
                            "precisOverlayCode": "frost",
                            "night": false,
                            "min": 1,
                            "max": 11
                        }
                    ]
                }
            ],
            "units": {
                "temperature": "c"
            },
            "issueDateTime": "2024-06-30 18:09:38"
        }
[... 190 lines cut ...]
float observational_observations_rainfall_since9AMAmount = observational_observations_rainfall["since9AMAmount"];

for (JsonPair observational_station : observational["stations"].as<JsonObject>()) {
  const char* observational_station_key = observational_station.key().c_str(); // "temperature", ...

  int observational_station_value_id = observational_station.value()["id"]; // 438, 438, 437, 438, 438, ...
  const char* observational_station_value_name = observational_station.value()["name"]; // "Viewbank", ...
  float observational_station_value_lat = observational_station.value()["lat"]; // -37.741, -37.741, ...
  float observational_station_value_lng = observational_station.value()["lng"]; // 145.097, 145.097, ...
  float observational_station_value_distance = observational_station.value()["distance"]; // 20.2, 20.2, ...

}

const char* observational_issueDateTime = observational["issueDateTime"]; // "2024-06-30 19:00:00"

JsonObject observational_units = observational["units"];
const char* observational_units_temperature = observational_units["temperature"]; // "c"
const char* observational_units_amount = observational_units["amount"]; // "mm"
const char* observational_units_speed = observational_units["speed"]; // "km/h"
const char* observational_units_distance = observational_units["distance"]; // "km"
const char* observational_units_pressure = observational_units["pressure"]; // "hPa"

If anyone can assist with my question, it would be greatly appreciated.

Thanks.

bblanchon, 2024-07-01:
Hi @justinpanta,

Here is how to get the weather forecast data for the first day:

```c++
JsonObject forecasts_weather_day_1 = doc["forecasts"]["weather"]["days"][0];
```

Best regards,
Benoit

justinpanta, 2024-07-02:
Thanks @bblanchon.

I can confirm that I can access the forecast data with the info you provided, however only within the for loop. Is there a way I can access these variables outside of the for loop?

bblanchon, 2024-07-04:
`doc["forecasts"]["weather"]["days"][0]` is independent of the loop and can be used outside of it.

justinpanta, 2024-07-04:
Yes, confirming I made an error in the code.

Thanks so much!

closed 2024-07-04 by justinpanta

bblanchon, 2024-07-05:
You're welcome, @justinpanta.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: