bblanchon/ArduinoJson #2108: Problem when a JSON from Firefox to my ESP8266

boggiz, 2024-07-01:
**Description**
When I send this JSON from Firefox to my ESP8266: http://192.168.69.143/PG/rec
http server returns <Not Found> if rec or PG are omitted => correct
on the other hand it systematically returns <deserializeJson() failed: EmptyInput> whatever follows the ?
http://192.168.69.143/PG/rec?key=25
Something is surely incorrect or missing in my script but what?

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns deserialization
4. `deserializeJson()` returns `EmptyInput`
5. Input comes neither from an HTTP response, nor a file, nor a serial port
6. Input type is a string
7. The first character is not NUL

**Environment**
* Microcontroller: ESP8266
* Core/Framework: NodeMCU1.0 (ESP-12E)
* IDE: Arduino IDE 1.8.19

**Reproduction code**
```c++
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ArduinoJson.h>

ESP8266WebServer server(80); // create the webserver on port 80

// setup the wifi
char ssid[] = "AC7-guests"; // your network SSID (name)
char pass[] = "Cyprianes28"; // your network password

void handleRoot() {
  server.send(200, "text/plain", "ESP8266 on line");
}

void setup() {
  Serial.begin(115200); // output to the serial terminal for debugging
  WiFi.begin(ssid, pass); //Connect to the WiFi network
  Serial.println(""); // print a new line
  Serial.print("Connecting .");
  while (WiFi.status() != WL_CONNECTED) {  //Wait for connection
    delay(500);
    Serial.print(".");
  }
  Serial.println(""); // print a new line

  server.on("/", handleRoot); //Associate the handler function to the path
  server.on("/PG/rec",HTTP_GET,recData);
  server.on("/PG/send", []() { //Define the handling function for the path
    server.send(200, "text/plain", "send");
  });
  
  server.begin(); //Start the server
  Serial.print("Server listening, Open ");
  Serial.print(WiFi.localIP());
  Serial.println(" in your browser.");
}

void recData(*char jsonString){
   StaticJsonDocument<300> JSONData;

   // Deserialize the JSON document
   String jsonString = server.arg("plain");
   DeserializationError error = deserializeJson(JSONData, jsonString);

  // Test if parsing succeeds.
  if (error) {
    Serial.println(jsonString);
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    server.send(500,"application/json","Error in parsing");
    return;
    }
  else{
   if(JSONData.containsKey("number")){
    server.send(200,"application/json",String(JSONData["number"].as<int>())+" Received");
   }
   else{
     Serial.println("Bad JSON");
     server.send(400,"application/json","Bad JSON");
   }
  }
}

void loop() {
  server.handleClient(); //Handling of incoming requests
}
```

bblanchon, 2024-07-02:
Hi @boggiz,

What is the content of the JSON input string?
Can you ~double~triple-check that it's not empty?

Best regards,
Benoit

boggiz, 2024-07-03:
Thank you for taking care of my problem @bblanchon .
Into Firefox url's I wrote: http://192.168.69.143/PG/rec?key=25
In a red banner Firefox print **SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data** 
ESP serial monitor print **EmptyInput**
After many tries I never managed to get any other error than Empty Input
Best regards 
Boggiz

bblanchon, 2024-07-10:
I'm closing this issue because it doesn't look like your problem is due to ArduinoJson.
Thank you for your understanding.

closed 2024-07-10 by bblanchon