bblanchon/ArduinoJson #2112: ArduinoJson 7.1.0 Compile error [ invalid use of array with unspecified bounds ]

gmgunderground, 2024-07-14:
**Description**
invalid use of array with unspecified bounds

.pio\libdeps\uno_r4_minima\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:22:19: error: invalid use of array with unspecified bounds
     return dest_->write(s, n);

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. The error is not in the list

**Environment**
* Microcontroller: Arduino R4 Minima
* Core/Framework: framework = arduino
* IDE: PlatformIO

**Reproduction code**
```c++
Only including 
#include <ArduinoJson.h>
I have also tryed with 
#include <ArduinoJson.hpp>
But in this case the error is 
src\main.cpp:18:1: error: 'JsonDocument' does not name a type
 JsonDocument doc;
```

gmgunderground, 2024-07-14:
I found the problem, as reported in documentation serializeJsonPretty acept char and i'm using "unsigned char" because I'm using a memory buffer
size_t serializeJsonPretty(const JsonDocument& doc, char output[size]);
from the mempry perspetice char and unsigned char are the same element, baut if I use (and is not rare to use a byte buffer in embedded) I can't serialize if I don't create a slow my custom writer

bblanchon, 2024-07-15:
Hi @gmgunderground,

I don't get any error with the provided sample code.
See this online demo: https://wandbox.org/permlink/Tu38w0wCx8SNfy7F

Please provide a sample code that reproduces the error.

Best regards,
Benoit

closed 2024-07-16 by gmgunderground