bblanchon/ArduinoJson #2114: queuing ArduinoJson objects (was: https://github.com/bblanchon/ArduinoJson/issues/308 )

mhaberler, 2024-07-22:
I found this interesting example of directly pushing JsonDocuments through a queue in the venerable [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway/tree/ada83bf227aadbcc09c572d2621ff629ef48220d) code base, and it uses the `JsonBundle`  approach outlined in #308

https://github.com/1technophile/OpenMQTTGateway/blob/ada83bf227aadbcc09c572d2621ff629ef48220d/main/main.ino#L88

https://github.com/1technophile/OpenMQTTGateway/blob/ada83bf227aadbcc09c572d2621ff629ef48220d/main/main.ino#L418-L433

https://github.com/1technophile/OpenMQTTGateway/blob/ada83bf227aadbcc09c572d2621ff629ef48220d/main/main.ino#L527-L529

do the changes from v6->v7 (StaticJsonDocument gone, all heap these days) suggest any change in the approach or are the examples from https://github.com/bblanchon/ArduinoJson/issues/308 still valid?

bblanchon, 2024-07-23:
Hi @mhaberler,

In short:

* `std::queue<JsonDocument>` => OK
* `std::queue<JsonObject>` => not OK

Best regards,
Benoit

mhaberler, 2024-07-23:
thanks a lot, appreciated!

best regards
Michael

bblanchon, 2024-07-24:
You're welcome, Michael.
Thank you for using ArduinoJson!

closed 2024-07-24 by bblanchon