bblanchon/ArduinoJson #2118: 'struct SpiRamAllocatorST' has no member named 'reallocate'; did you mean 'deallocate'?

ghmpi, 2024-08-01:
Hello,

I'm having a bit of trouble with a project. I need to turn on PSRAM for other reasons on this board, but when I do, I can't compile. I get the error noted below. Works fine when PSRAM is disabled. Couldn't find anything relevant using the troubleshooter or searches. Any guidance is appreciated.

Thank you!

Regards,
-Moses

ArduinoJson v7.1.0
Arduino IDE 1.8.19
ESP32-WROVER-E module, original devkit-c board

Arduino\libraries\ArduinoJson\src/ArduinoJson/compatibility.hpp: In instantiation of 'void* ArduinoJson::V710PB22::detail::AllocatorAdapter<TAllocator>::reallocate(void*, size_t) [with TAllocator = SpiRamAllocatorST; size_t = unsigned int]':
Arduino\libraries\ArduinoJson\src/ArduinoJson/compatibility.hpp:88:9:   required from here
Arduino\libraries\ArduinoJson\src/ArduinoJson/compatibility.hpp:89:23: error: 'struct SpiRamAllocatorST' has no member named 'reallocate'; did you mean 'deallocate'?
     return _allocator.reallocate(ptr, new_size);
            ~~~~~~~~~~~^~~~~~~~~~
            deallocate

bblanchon, 2024-08-02:
Hi Moses,

In ArduinoJson 6, the `reallocate()` member was optional for custom allocator classes.
It's now mandatory in ArduinoJson 7.

You must update the `SpiRamAllocatorST` like so:

```diff
  struct SpiRamAllocatorST {
   void* allocate(size_t size) {
     return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
   }
   
   void deallocate(void* pointer) {
     heap_caps_free(pointer);
   }

+  void* reallocate(void* ptr, size_t new_size) {
+    return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);
+  }
  };
```

This example comes from the documentation, but your version could be different, so please adapt accordingly.

See also the [guide](https://arduinojson.org/v7/how-to/upgrade-from-v6/) to upgrade your code to ArduinoJson 7.

Best regards,
Benoit

ghmpi, 2024-08-03:
Benoit,

Thank you very much for the quick reply and examples.

I'm not very well versed here, so please excuse me if this is a silly question. But where would I add the void* reallocate ?

+  void* reallocate(void* ptr, size_t new_size) {
+    return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);
+  }

Thank you again!

Regards,
-Moses

bblanchon, 2024-08-04:
As shown in my previous message, you must add the definition of `reallocate()` in the definition of `SpiRamAllocatorST`.

Alternatively, you can downgrade to ArduinoJson 6, which doesn't require this function.

ghmpi, 2024-08-05:
Benoit,

Thank you, that worked. I added the function to SpiRamAllocatorST

One curious thing to note is that I had to add it to another Libraries files! Looks like he had to do some wrapping to make his code work too.
\Arduino\libraries\AutoConnect\src\AutoConnectJsonDefs.h

A snippet of code from the .h file below if you are curious.

Thanks again!

Regards,
-Moses

`/**
 * Wrapping definition to ensure version compatibility of ArduinoJson.
 * @file AutoConnectJsonDefs.h
 * @author hieromon@gmail.com
 * @version  1.3.1
 * @date 2021-10-03
 * @copyright  MIT license.
 */

#ifndef _AUTOCONNECTJSONDEFS_H_
#define _AUTOCONNECTJSONDEFS_H_

#include <ArduinoJson.h>

/**
 * Make the Json types and functions consistent with the ArduinoJson
 * version. These declarations share the following type definitions:
 * - Difference between reference and proxy of JsonObject and JsonArray.
 * - Difference of check whether the parsing succeeded or not.
 * - The print function name difference.
 * - The buffer class difference.
 * - When PSRAM present, enables the buffer allocation it with ESP32 and
 *   supported version.
 */`

bblanchon, 2024-08-05:
Hi Moses,

Congratulations on making this work!
Please send a Pull Request to the AutoConnect library so that other users can benefit from this.
If you don't feel comfortable with creating a Pull Request, at least open an issue there and link to the current issue; they will know what to do.

Best regards,
Benoit

closed 2024-08-05 by bblanchon