bblanchon/ArduinoJson #2119: Some trouble using variables as element names in JsonDocument

ghmpi, 2024-08-05:
Hello,

I have a project using ArduinoJson grabbing weather data for a display. All is working well expect that I would love to make it user configurable from variables, in case the weather host needs to be changed. Let me explain.

I have "doc" as a regular JsonDocument, and the function that grabs weather data returns the current info using the following:
return doc[features][1][properties][rawOb];

I would like to make the fields user configurable and hence the values for the JsonDocument are coming from Strings. For example..

  String DOC1 = "features";
  String DOC2 = "1";
  String DOC3 = "properties";
  String DOC4 = "rawOb";
  return doc[DOC1][DOC2.toInt()][DOC3][DOC4];

This works save for a few issues.. you cannot specify a string where doc expects an INT, understandable, and the DOC2.toInt() portion fixes this. The problem becomes that I cannot know at compile time what user fields will be INTs and what are Strings. Applying toInt() to all the variables breaks the return, I was hoping that would do it!

Another issue is I cannot know beforehand how many elements the user provides in the config, in the example above there are 4, but there may be 1 or upwards of about 5-8 or so. I tried filling in empty elements with NULLs.. but this did not work.

There may be elaborate ways of getting what I want, I'm still racking my brain to see what can be done. But is anyone aware of an elegant / simple method to accomplish this?

I've spent several hours scouring posts and reading documentation but so far I've come up with nothing useful. I apologize if this all sounds silly, maybe I'm just not thinking of the solution correctly.

Info on my setup below.  I appreciate the assistance.

Regards,
-Moses

ArduinoJson v7.1.0
Arduino IDE 1.8.19
ESP32-WROVER-E module, original devkit-c board

bblanchon, 2024-08-06:
Hi Moses,

Since ArduinoJson 7.1.0, you can use a `JsonVariant` as a key or an index, and it is resolved at runtime (issue #2080).

Best regards,
Benoit

ghmpi, 2024-08-06:
Benoit,

I don't think I have the same problem as in issue 2080. I'll give a simple example below.

My closest fix I can think of right now is to loop through each user provided element, test and provide the proper string/INT, grab the resulting Json, reassign it to another jsondocument and reiterate the loop until the user provided elements are done. But this seems messy.

Appreciate the assistance.

Regards,
-Moses

Given the following, which is provided by the USER and can change
String DOC1 = "features";
String DOC2 = "1";
String DOC3 = "properties";
String DOC4 = "rawOb";

// The following does not work..
return doc[DOC1][DOC2][DOC3][DOC4];

// Any numeric value needs to be converter, see DOC2 below, but since these are user provided I don't know which ones will be numeric.
return doc[DOC1][DOC2.toInt()][DOC3][DOC4];

bblanchon, 2024-08-09:
Try to use a JSON document to store the user configuration, so you can leverage JsonVariant's ability to converter implicitly to string or integer.

For example, the configuration could look like this:

```json
{
  "path": [
    "features",
    1,
    "properties",
    "rawOb"
  ]
}
```

ghmpi, 2024-08-19:
Benoit,

Thank you for the advice. I really do not want to add another layer of complexity to my configuration, but may have to try your suggestion as I have found no other easier way of making this work. Wish just providing strings worked! :)

Regards,
-Moses

closed 2024-10-15 by bblanchon