bblanchon/ArduinoJson #2120: cppcheck: failed to expand 'ARDUINOJSON_BEGIN_PUBLIC_NAMESPACE'

robsonos, 2024-08-16:
Hi,

Just reporting this issue while running `cppcheck`:

`...ArduinoJson/src/ArduinoJson/Polyfills/preprocessor.hpp:7: [high:error] failed to expand 'ARDUINOJSON_BEGIN_PUBLIC_NAMESPACE', Invalid ## usage when expanding 'ARDUINOJSON_CONCAT_': Unexpected token ')' [preprocessorErrorDirective]`

Cheers

marco86to, 2024-08-18:
Hi, I have found the same bug. 
Cheers

bblanchon, 2024-08-21:
Hi @robsonos and @marco86to,

Here is line 7 of `Polyfills/preprocessor.hpp`:

```c++
#define ARDUINOJSON_CONCAT_(A, B) A##B
```

There is no error here!

Also, I ran cppcheck 2.10 on this file and got no error.

Best regards,
Benoit

robsonos, 2024-08-22:
Hi @bblanchon,

For more context, I am using `platformio/tool-cppcheck` from PlatformIO.
I did some testing and coundn't find any issues with the macro expansions of `ARDUINOJSON_VERSION_NAMESPACE` and `ARDUINOJSON_CONCAT_`. I wonder if the issue is related to `cppcheck` itself like in https://community.platformio.org/t/wrong-issues-from-cppcheck/18410/7.
In terms of PlatformIO itself, the following will suppress the error:

```
check_flags =
    cppcheck: --suppress=*:*.pio\* 
```

closed 2024-10-15 by bblanchon