bblanchon/ArduinoJson #2121: containsValue() method

julianvitor, 2024-08-22:
The ArduinoJson library currently has a method containsKey() to check if a key exists in a JsonDocument, but there isn’t a method to check if a specific value is present.

Please add a containsValue() method to the ArduinoJson library. This method would help in checking if a specific value exists anywhere within a JsonDocument, making it easier to handle JSON data.

Why It’s Useful:

Simplifies Code: You won’t need to manually search through all key-value pairs.
Efficient Searches: Allows for quicker checks if a value is present in large JSON documents

bblanchon, 2024-08-23:
Hi @julianvitor,

I don't think searching for values in an object is such a common use case that it requires a built-in function.
Instead, you can easily define one out of the library like so:

```c++
template<typename T>
bool containsKey(JsonObjectConst obj, T value) {
  for (auto kvp : obj) {
    if (kvp.value() == value)
      return true;
  }
  return false;
}
```

BTW, I plan on removing `containsKey()` in v8 since there are better alternatives, such as `doc["key"].is<int>()`, which not only checks that the key exists but also validates the type.

Best regards,
Benoit

julianvitor, 2024-08-26:
`doc["key"].is<int>()` is a great idea.

closed 2024-08-26 by julianvitor