bblanchon/ArduinoJson #2122: Keep a JsonDocument in a PSRAM  - Container

johnnytolengo, 2024-08-26:
Microconroller: ESP32 with  4MB Ram
ArduinoJson version: v7.1.0
Framework: Arduino Espressif 32 - 6.5.0

I am looking a way to keep a JsonArray/JsonDocument in PSRAM and update it as the data  increase or decrease. Actually my application is a kind of queue for Mqtt I add data(json format)  to a JsonArray when a new value arrives and I delete that data once it's sent.  
SPIFF is not a candidate because of it's limited read an write cycles.

1. first try:
```
 char* _mqtt_buff = ( char*) calloc (D_MEGABYTE, sizeof(char));  // Allocate 1MB
//Serialize data to char *
serializeMsgPack(some_jsonDoc, _mqtt_buff);  //that's not working
```

Another idea could be to allocate std::vector in PSRAM but I have not idea how it's that possible.

Probably it's there some more effective ways on how to do that ArduinoJson container.
I'll appreciate any idea.

bblanchon, 2024-08-27:
Hi @johnnytolengo,

What do you mean by "that's not working"?
Did you try the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/)?

Best regards,
Benoit

johnnytolengo, 2024-08-27:
@bblanchon I found a good solution using a custom struct + xQueueCreate.

`QueueHandle_t _mqtt_queue = xQueueCreate( 1000, sizeof( struct mqtt_queue_struct ) ); //allocated in PSRAM by default
`
It's working very well an seems to be very safe even using multiple threads.

How is possible to save a JsonDocument in a std::vector<uint8_t> &data; ?  I don't know if it's that possible, if yes how to put back the vector data in a JsonDocument?

Thanks.

bblanchon, 2024-08-31:
ArduinoJson has no built-in support for `std::vector`.
Instead of a `std::vector`, why don't you use a `std::string`?

closed 2024-10-15 by bblanchon