bblanchon/ArduinoJson #2124: Library fails to parse a hybrid of single and double quotes

zvirja, 2024-09-16:
**Describe the bug**  
Library fails to parse JSON when there is a mix of single and double quotes. That happens when using single quotes for JSON structure.

I know that it's a bit weird to use single quote for JSON elements, but that's what HomeAssistant & EspHome are doing 😢 

**Troubleshooter report**  
Irrelevent

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266, ESP32
* Core/runtime: ESPHome

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
auto json = R"({ 'summary': 'x "a\'b" y' })"
          
ArduinoJson::JsonDocument doc;
auto res = ArduinoJson::deserializeJson(doc, json);
auto summary = doc.as<ArduinoJson::JsonObject>()["summary"].as<std::string>();
```

**Program output**  

Expected output:

```
res: OK
summary: x "a\'b" y
```

Actual output:

```
res: InvalidInput
summary: null
```

bblanchon, 2024-09-17:
Thank you very much for reporting this issue.
The fix will be included in ArduinoJson 7.2.0 very soon.

closed 2024-09-17 by bblanchon

zvirja, 2024-09-17:
@bblanchon Wow, thanks for a really fast fix! 😯☺

Is it worth to have logic that escape sequences are different depending on whether we are in single or double quotes context? So that in the `{ "summary": "a\\'b" }` context sequence is kept as-is and vice versa?

bblanchon, 2024-09-17:
This one should work as before since it's the backslash that is escaped, not the quote.

```json
{"summary": "a\\'b"}
```

However, the following one used to return `InvalidInput` but will now return `Ok`:

```json
{"summary": "a\'b"}
```

I think it's correct, even if it's not standard.
ArduinoJson never intended to be a *strict* parser.

bblanchon, 2024-09-18:
The fix is available in [ArduinoJson 7.2.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.2.0).

zvirja, 2024-09-18:
@bblanchon Just verified it with Home Assistant - works like a charm! Thanks for your fast reply and being so nice 😊

commit f806a42c (2024-09-17): Add support for escape sequence `\'`

Fixes #2124