bblanchon/ArduinoJson #2125: Value is always zero

rwb196884, 2024-09-17:
```
    //JsonDocument jDoc; // This is the new way of doing it in v7 and it doesn't work; applied is alyways zero.
    StaticJsonDocument < 64 > jDoc;
    if (wifi_client.available()) {
      deserializeJson(jDoc, wifi_client);
      Serial.println("RECEIVED");
      serializeJson(jDoc, Serial);
      temp = jDoc["temp"];
      unsigned long applied = jDoc["applied"];
      Serial.print(t_current);
      Serial.print(" -> ");
      Serial.println(temp);
      Serial.print("applied: ");
      Serial.println(applied);
```
prints
```
RECEIVED
{"temp":60,"applied":39543}0 -> 60
applied: 39543
```
but with `JsonDocument` it prints
```
RECEIVED
{"temp":60,"applied":39543}0 -> 60
applied: 0
```

bblanchon, 2024-09-18:
Hi @rwb196884,

Do you mean that the code works with ArduinoJson 6 but not with ArduinoJson 7, or that it works with ArduinoJson 7 but only when using `StaticJsonDocument`?

Best regards,
Benoit

rwb196884, 2024-09-19:
Version 7 is installed. 

This is the latest fuckup.
```
      StaticJsonDocument<64> jDocFlow;
      jDocFlow["t_flow"] = t_flow;
      jDocFlow["t_return"] = t_return;
      jDocFlow["foo"] = -1;

      char payload[64];
      serializeJson(jDocFlow, payload);
      bool p = false;
      while(!p) {
        connect_wifi();
        connect_ap();
        connect_mqtt();
        p = mqtt_client.publish(mqtt_topic_temp, payload, true);
        if (!p) {
          blink(6);
        } // Error 6: couldn't report temperatures to MQTT.
        else {
          Serial.println("--published temp--");
          serializeJson(jDocFlow, Serial);
          Serial.println();
        }
      }
```
outputs
```
--published temp--
{}
```

bblanchon, 2024-09-20:
What does the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/) say?

rwb196884, 2024-09-24:
Getting some junk from `Serial.print`
```
Reported temperatures: {"tt�low": 28, "t_re�r!�
```
I wonder if the board is fucked?

bblanchon, 2024-09-26:
This last issue suggests a stack overflow.
Reduce the size of the stack variables.
Start by removing `char payload[128]`.

rwb196884, 2024-10-10:
How do you determine whether something is stored in the heap on on the stack?

I imagine it's something to do with where you declare the variable -- either inside a function of outside of any function -- but I can't find any documentation.

bblanchon, 2024-10-13:
There is no quick answer to this question, unfortunately.
I invite you to check out the "Missing C++ Course" in my [book](https://arduinojson.org/book/) or any other C++ course.

closed 2024-10-30 by bblanchon