bblanchon/ArduinoJson #2127: `containsKey()` alternative?

vortigont, 2024-09-19:
In v7.2 `containsKey()` is no longer available, but is there any other way to figure out if a k:v pair exist in document if you do not know/do not care about the type of value?
I.e. I have a code like this:
```
    if (doc.containsKey(key)){
        doc.remove(key);
        serializeJson(doc, destination);
    }
```

the `containsKey()` call here is to avoid serializing/saving doc if `remove()` call was blank and doc has not changed.

I could iterate the keys and do compare key name with name-to-delete variable, but this does not look nice here.
Maybe `remove()` could return boolean result of it's operation?

Thanks!

sblantipodi, 2024-09-19:
+1 fpr this question, I sincerely don't understand why containsKey() has been removed, 
it broke a lot of things specially when you are retrieving informations when you don't know the type.

I have objects that may contain a String or an int like this "50" or 50. 
I then use the .as<int>() to use it as an integer. 

But without containsKey how can I know if the value is present or not?

how can I check now if color is set in an object like this?

```
{
  "color" : {
    "r" : 255,
    "g" : 82,
    "b" : 0,
    "colorMode" : 3,
    "colorOrder" : 1
  }
}
```

I think that this deprecations broke too much things and requires too much effort for the switch.
Even with a lot of effort some errors may occur and a full regression test is needed.

I would expect such a big impact on a new major version of the lib, not in a small minor version update.

sblantipodi, 2024-09-19:
Hi answer to my previous comment and to the OP.
Knowing that Benoît is the master of the gurus, there is always an answer behind his decisions.

So, I searched for an answer in his docs, that most of the time has an answer to your questons.
and here it is:
https://arduinojson.org/v7/api/jsondocument/containskey/

simply use:
```
// Before
if (doc.containsKey("value")) {
  int value = doc["value"];
  // ...
}

// After
if (doc["value"].is<JsonVariant>()) {
  int value = doc["value"];
  // ...
}
```

bblanchon, 2024-09-19:
`containsKey()` is still available, only deprecated, so your existing code should still compile.

In most cases, you can replace `doc.containsKey(key)` with `doc[key] != nullptr`, but if you must accept `null` values too, then you must use `doc[key].is<JsonVariant>()`.

As I explained in [the blog post](https://arduinojson.org/news/2024/09/18/arduinojson-7-2/), the motivation behind this change is to make an API that is easy to use correctly and hard to use incorrectly (which seems to work pretty well :wink:).

vortigont, 2024-09-19:
Oh! I did not know about `doc[key].is<JsonVariant>()` trick! This completely solves all the cases including `null` and `0`.
I think this trick should be specifically mentioned in [that blog post](https://arduinojson.org/news/2024/09/18/arduinojson-7-2/).

closed 2024-09-19 by vortigont

bblanchon, 2024-09-19:
I added it this morning after seeing your message.

DaeMonSxy, 2024-10-06:
Hi all, 
I bean dealing with int or bool variables, so had troubles with checking 0-1 also.
come across this solution, however, does not seem to be working right.

`bool ifKey(const JsonDocument &doc, const char *key)
{
    if (doc[key].is<JsonVariant>()) // does not exists if received 0 or 1 
    //if (doc[key].is<int>()) // works good with 0 or 1
    {
        Serial.println(" - exists  ");
        return true;
    } else {
        //Serial.print(key);
        //Serial.println("- does not exists");
        return false;
    }
}`

did i missed something? .is<JsonVariant>() expected to replace containsKey, regardless of 0,1 or "string" ?

bblanchon, 2024-10-07:
@DaeMonSxy, the canonical way to test for `int` or `bool` is to test both types: `var.is<int>() || var.is<bool>()`.

In your case, testing for `is<JsonVariant>()` doesn't work because it always returns `false` for a `const JsonDocument&`.
Indeed, a read-only `JsonDocument` will not let you get a *mutable* reference to a member.
You can fix this code by calling `is<JsonVariantConst>()` instead.