bblanchon/ArduinoJson #2131: UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in idf_component.yml

wcjona, 2024-09-25:
**Describe the bug**  
UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in idf_component.yml due to non-UTF-8 character is present. 

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32S3
* Core/runtime: ESP32 w/ Ardunio Core Lib
* IDE: VSCode + ESP-IDF

**Reproduction**  
Build main.cpp with ArduinoJson Lib

```c++
#include <ArduinoJson.h>
extern "C" void app_main() {
}
```

**Compiler output**    
-- Building ESP-IDF components for target esp32
CMake Error at C:/Users/Jonat/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:552 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>      main()
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 130, in main   
      args.func(args)
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 29, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 98, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_manager\core.py", line 753, in prepare_dep_dirs
      ).load()
        ^^^^^^
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 154, in load
      if self.is_valid:
         ^^^^^^^^^^^^^
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 139, in is_valid
      self.validate()
    File "C:\Users\Jonat\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\idf_component_tools\manager.py", line 76, in validate
      manifest_dict = yaml.safe_load(self.path.read_text()) or {}
                                     ^^^^^^^^^^^^^^^^^^^^^
    File "pathlib.py", line 1059, in read_text
    File "encodings\cp1252.py", line 23, in decode

  UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 96:
  character maps to <undefined>

Call Stack (most recent call first):
  C:/Users/Jonat/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:7 (project)

wcjona, 2024-09-25:
I'll write up a pr for this :)

bblanchon, 2024-09-26:
Thank you very much for this.
I know you wanted to write a PR, but this was just a one-second fix, so I did it myself.

commit 25129936 (2024-09-26): Replace problematic symbol in `idf_component.yml`

Fixes #2131
commit 617c0329 (2024-09-26): Replace problematic symbols in `idf_component.yml`

Fixes #2131
closed 2024-09-26 by bblanchon