bblanchon/ArduinoJson #2132: Calculating the buffer size

butaikis, 2024-10-05:
Hi. I sometimes use an [assistant](https://arduinojson.org/v7/assistant/#/step2) for approximate calculation of the JSON buffer and there is a question. There is RAM Usage on the page, there is the main value and just below it is Peak. In which cases is the Peak value reached?

bblanchon, 2024-10-06:
Hi @butaikis,

Memory consumption increases temporarily during deserialization; that's when the peak is reached.

In other words, the "regular" RAM usage corresponds to the final `JsonDocument`, whereas the peak accounts for the temporary overhead of `deserializeJson()`.

Best regards,
Benoit

butaikis, 2024-10-06:
@bblanchon Understood, thanks for the explanation. Please tell me if I use the reading from LittleFS function.
```
  DynamicJsonDocument BUFFER_JSON(61440);
  File FRead= FS_User.open("/test.json", "r");
  DeserializationError error = deserializeJson(BUFFER_JSON, FRead);
  FRead.close();
```
 Is it more correct to choose Input: Stream in the [assistant](https://arduinojson.org/v7/assistant/#/step1)?

bblanchon, 2024-10-07:
LittleFS's `File` implements the `Stream` interface; that's why you can pass it directly to [`deserializeJson()`](https://arduinojson.org/v7/api/json/deserializejson/).
In the [Assistant](https://arduinojson.org/v7/assistant/), you should, therefore, choose "Stream" for the input type.

By the way, in ArduinoJson 7, you can simply declare your document as `JsonDocument BUFFER_JSON;`.

butaikis, 2024-10-07:
@bblanchon Thanks!

closed 2024-10-07 by butaikis

bblanchon, 2024-10-07:
You're welcome, @butaikis.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: