bblanchon/ArduinoJson #2138: array in an array not serializing

got879, 2024-10-26:
**Description**
nothing is serialized or sent
MqttClient max payload is set to 65535
overflow = 0

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns serialization
4. Output is incomplete
5. `JsonDocument::overflowed()` returns `false`
6. The empty or missing value is an array
7. Calling `add<JsonArray>()` or `to<JsonArray>()` does not fix the issue
8. The output is a `MqttClient`
9. Increasing the buffer size does not solve the issue

**Environment**
* Microcontroller: SAMD21 (nano 33 IoT)
* Core/Framework: ESP32 (ublox NINA module), WifiNINA 1.50
* IDE: Arduino IDE 1.18.12

**Reproduction code**
```c++
JsonArray data = doc.add<JsonArray>();
  for ( int t = 0; t < dataCount; t++ ) {
    data.add(runSensors[t]);
    data.add(runTimes[t]);
  }
```

**Remarks**
Sending to mosquitto broker on digital ocean - all is working except data message is not. In a previous version of the code it serialized but sending the data message crashed the Wifi connection after a few cycles.

bblanchon, 2024-10-27:
Hi @got879,

By definition, `doc.add<JsonArray>()` appends a new empty array to the root array; so, after the first line, `doc` contains:

```json
[[]]
```

Then, `data.add()` appends new values to the inner array, so you should get something like this:

```json
[[1.2,1730029927,2.3,1730029928]]
```

If, instead, you expect the output to be:

```json
[1.2,1730029927,2.3,1730029928]
```

you need to replace `doc.add<JsonArray>()` with `doc.to<JsonArray>()`.

I don't see anything else suspicious in the reproduction code, so if you need more assistance, please provide more context. In particular, please include the output of `serializeJson(doc, Serial)`.

Best regards,
Benoit

got879, 2024-10-28:
Got it working now. Thank you.

One more issue. Previously using v6 using  I have the data message formatted as follows using `JsonArray d = data.createNestedArray();`:

[20,11459,13],[20,11461,15]

where 20 is a sensor number, 11459 is an elapsed time, and 15 is distance

With the current v7 code the message output is:

[18,12579,15,0,0]

There are three distance values nested for a single sensor.

Thanks

bblanchon, 2024-10-30:
Hi @got879,

The equivalent of `createNestedArray()` is `add<JsonArray>()`, for the reasons explained in my previous message.

Did you know you could use the [ArduinoJson Assistant](https://arduinojson.org/v7/assistant/) to generate a sample program?
In your case, it produces:

```c++
// Stream& output;

JsonDocument doc;

JsonArray doc_0 = doc.add<JsonArray>();
doc_0.add(20);
doc_0.add(11459);
doc_0.add(13);

JsonArray doc_1 = doc.add<JsonArray>();
doc_1.add(20);
doc_1.add(11461);
doc_1.add(15);

doc.shrinkToFit();  // optional

serializeJson(doc, output);
```

Best regards,
Benoit

closed 2024-11-20 by bblanchon