bblanchon/ArduinoJson #2139: Pb with filter : if i use string value it's ok, but with char[] : just the first value of filter ....

StefT09, 2024-11-02:
The input (payload) is a string, lenght : 20000

```
          StaticJsonDocument<64> filter;
          JsonObject filter_values = filter.createNestedObject("values");
          filter_values["2024-10-31"] = true;
          filter_values["2024-11-01"] = true;
          filter_values["2024-11-02"] = true;
          StaticJsonDocument<192> doc;
          DeserializationError error = deserializeJson(doc, payload, DeserializationOption::Filter(filter));
```

with 

`          serializeJsonPretty(doc, Serial);`

the result is ok :
{
  "values": {
    "2024-11-02": "BLUE"
    "2024-11-01": "BLUE"
    "2024-10-31": "BLUE"
  }
}

But i need variable filter ... I try :

```
char rteHier[11] = { 0 };
char rteJour[11] = { 0 };
char rteDemain[11] = { 0 };

...

sprintf(rteJour, "%i-%02d-%02d", year(jourRte), month(jourRte), day(jourRte));
sprintf(rteHier, "%i-%02d-%02d", year(jourRte - SECS_PER_DAY), month(jourRte - SECS_PER_DAY), day(jourRte - SECS_PER_DAY));
sprintf(rteDemain, "%i-%02d-%02d", year(jourRte + SECS_PER_DAY), month(jourRte + SECS_PER_DAY), day(jourRte + SECS_PER_DAY));

...

StaticJsonDocument<64> filter;
JsonObject filter_values = filter.createNestedObject("values");
filter_values[rteJour] = true;
filter_values[rteHier] = true;
filter_values[rteDemain] = true;
StaticJsonDocument<192> doc;
DeserializationError error = deserializeJson(doc, payload, DeserializationOption::Filter(filter));
```

the result is not ok :
{
  "values": {
    "2024-11-02": "BLUE"
  }
}

Only the first value of 'filter_values' is returned ... if i try 
```
filter_values[rteHier] = true;
filter_values[rteJour] = true;
filter_values[rteDemain] = true;
```

i have : 
{
  "values": {
    "2024-11-01": "BLUE"
  }
}

* Microconroller: ESP8266
* IDE: Arduino IDE 2.3.3
* ArduinoJson 6.21.5

bblanchon, 2024-11-03:
Hi @StefT09,

Adding a string literal to a `JsonDocument` requires no additional memory, but adding a `char[]` makes a copy.
Therefore, you must increase the capacity of the filter.

As usual, use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the capacity.
Since the keys are `char[]`, you uncheck "Assume keys are `const char*`".
For example, if the filter contains three dates, the Assistant recommends a capacity of 128.

Alternatively, you can upgrade to ArduinoJson 7, which manages the memory automatically.

Best regards,
Benoit

StefT09, 2024-11-03:
Thank you Benoit !
it work perfectly 👍 !
I admit that I am absolutely a beginner in C C++ Arduino, being used to a language like VBA, the notions of string, char, pointers etc are rather obscure to me !
Thanks again for your response and for the work you have done and made available to everyone.

closed 2024-11-03 by StefT09