bblanchon/ArduinoJson #2140: garbage output when add objects to JsonArray in a loop

deepak-coding-art, 2024-11-06:
I have this code in a `ESPAsyncWebServer` route handler when i add the file info in the array in a loop i get this garbage data only happens in the loop

code:
```cpp
 File dirFile = LittleFS.open(dir);
    // Get all files in this dir
    JsonDocument jsonData;
    JsonArray filesArray = jsonData.to<JsonArray>();

    File file = dirFile.openNextFile();
    LOG_DEBUG("------------------------- START -------------------------");
    while (file)
    {
        String log_string = "File: " + String(file.name()) + ", Size: " + String(file.size()) + ", IsDir: " + String(file.isDirectory());
        LOG_DEBUG(log_string);
        JsonObject fileData = filesArray.add<JsonObject>();
        fileData["name"] = file.name();
        fileData["size"] = file.size();
        fileData["is_dir"] = file.isDirectory();
        file.close();
        file = dirFile.openNextFile();
    }
    // For serialization in v7:
    String jsonString;
    serializeJsonPretty(jsonData, jsonString); // or use serializeJson for compact output

    LOG_DEBUG("------------------------- END -------------------------");
    LOG_DEBUG(jsonString);
    dirFile.close();
```

Output jsonData:
```txt
[
  {
    "name": "",
    "size": 15406,
    "is_dir": false
  },
  {
    "name": "��?4��?/littlefs/www/update",
    "size": 0,
    "is_dir": true
  },
  {
    "name": "update",
    "size": 2288,
    "is_dir": false
  },
  {
    "name": "styles",
    "size": 0,
    "is_dir": true
  },
  {
    "name": "\n    \"is_dir\": false\r\n  },\r\n  {\r\n    \"name\": \"styles\",\r\n    \"size\": 0,\r\n    \"is_dir\": true\r\n  },\r\n  {",
    "size": 0,
    "is_dir": true
  }
]
```

I am new with this can someone help with this

bblanchon, 2024-11-07:
Please try the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/).

deepak-coding-art, 2024-11-07:
I tried ArduinoJson Troubleshooter but can not figure out what is the problem.

deepak-coding-art, 2024-11-07:
I figured it out turns out the `file.name()` returns `const char *` so i converted it to string fist then added to the `JsonObject`
updated line:
```cpp
fileData["name"] = String(file.name());
```

closed 2024-11-07 by deepak-coding-art