bblanchon/ArduinoJson #2141: erro to save file on esp32

eraldo2312, 2024-11-18:
**Description**
when i try to save data on .txt file, in json format, the output from my esp32 send me the values that I expect to receive from a reading, but when I restart the esp to just read, the return is empty input. 
Can someone help me with this issue? Thanks.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns deserialization
4. `deserializeJson()` returns `EmptyInput`
5. Input comes from a file
6. The file is not the problem

**Environment**
* Microcontroller: esp32
* Core/Framework: esp idf
* IDE: vscode

**Reproduction code**
```c++
esp_err_t FileSystem::readFileJson(std::string fileName, const char* key, std::string& outValue) {
    FILE* file = fopen(fileName.c_str(), "r");
    if (!file) {
        // ESP_LOGE(TAG, "Failed to open file: %s", fileName);
        std::cout << "Failed to open file for reading: " << fileName <<std::endl;
        return ESP_FAIL; 
    }

    FileAdapter adapter(file); // Usa o adaptador para o arquivo
    // DynamicJsonDocument doc(jsonDocSize);
    JsonDocument doc;
    DeserializationError error = deserializeJson(doc, adapter);
    // fflush(file);
    // fclose(file); // Fechar o arquivo após a leitura

    // std::cout << "Key =  " << doc[key] <<std::endl;

    if(doc.isNull()){
        std::cout << "O documento está vazio" << std::endl;
    }

    std::cout << "Size = " << doc.size() << std::endl;

    if (error) {
        // ESP_LOGE(TAG, "Error deserializing JSON: %s", error.c_str());
        std::cout << "Error deserializing JSON: " << error.c_str() <<std::endl;
        return ESP_FAIL; 

    // JsonObject obj = doc.as<JsonObject>();
    if (doc[key].is<const char*>()) {
    // if (doc.containsKey(key)) {
        outValue = doc[key].as<std::string>();
        // ESP_LOGI(TAG, "Chave = %s, Valor = %s", key, outValue.c_str());
        std::cout << "key =  " << key << " value = " << outValue.c_str() <<std::endl;
        return ESP_OK; 
    } else {
        std::cout << "Key not found: " << key <<std::endl;
        return ESP_FAIL; //
    }
    fclose(file);
}

esp_err_t FileSystem::writeFileJson(std::string fileName, const char* key, const char* value) {
    FILE* file = fopen(fileName.c_str(), "w");
    if (!file) {
        // ESP_LOGE(TAG, "Failed to open file for writing: %s", fileName);
        std::cout << "Failed to open file for writing: " << fileName <<std::endl;
        return ESP_FAIL; 
    }
    
    FileWriter writer(file); 
    // DynamicJsonDocument doc(1024); 
    JsonDocument doc;
    // doc.clear();
    doc[key] = value; 
    // serializeJson(doc, writer);
    if (serializeJson(doc, writer) == 0) { 
        // ESP_LOGE(TAG, "Failed to write to file: %s", fileName);
        std::cout << "Failed to write to file: " << fileName <<std::endl;
        fclose(file);
        return ESP_FAIL;
    }
    fflush(file);
    vTaskDelay(10 / portTICK_PERIOD_MS);
    fclose(file); 
    return ESP_OK; 
}
```

bblanchon, 2024-11-20:
Hi @eraldo2312,

Please clarify: 

1. is the file empty after calling `writeFileJson()`?
2. is the file empty after reboot?

Best regards,
Benoit

eraldo2312, 2024-11-20:
hi Bblanchon/Arduinojson,
The file used to be empty after a reboot. Working here on the spiffs and
littlefs, probably the error was in my flash system, cuz now it's working
fine.
Thanks for your attention.

bblanchon, 2024-11-21:
You're welcome.
Thank you for the update.