bblanchon/ArduinoJson #2142: Change code from v6.19 to v7.latest

hasenradball, 2024-11-25:
Hi,

having changed the lib from actual v6.19 to v7.latest and see some points to update.
is there a general example how to setup code for Arduino JSON for v7. to be able to do a review with v7 in mind?

hasenradball, 2024-11-25:
Hi Benoit,

watched your video, great stuff.

I want to buy the paperback version!
Do you have one for me?

What is your suggestion working on the ESP-01?
Stay on 6.19.4 or switch to 7?

bblanchon, 2024-11-26:
Hi @hasenradball,

I'm overwhelmed these days, so I don't know when I'll be able to work on the paperback version.

For every 32-bit microcontroller, including ESP-01, I strongly recommend switching to ArduinoJson 7.

Best regards,
Benoit

hasenradball, 2024-11-26:
I replace the above code by this.

It would be nice if you can give a final confirm ;-)

```
void sendMessageJson(int id, bool state_relais, bool state_tor) {
  char json_str[JSON_MAX_OUTPUT_SIZE];
  // v6.x
  // Reserve DynamicJsonDocument doc(500); --> Heap
  // Reserve StaticJsonDocument<500> doc; --> Stack
  // v7.x
  JsonDocument doc;

  // Uhrzeit
  char timestrbuffer[6];
  uhr.get_timeformatted(timestrbuffer, 6, "%R");
  
  // Betriebszeit
  char durationstrbuffer[29];
  snprintf(durationstrbuffer, 29, "%02lld:%02u:%02u", esp_duration.get_hh(), esp_duration.get_mm(), esp_duration.get_ss());
 
  doc["Version"] = webIf.version;
  doc["Relais"] = state_relais;
  doc["Tor_offen"] = state_tor;
  doc["Temperature"] = static_cast<float>(ESP.getFreeHeap());
  //doc["Temperature"] = sensor_BME280.Data.Messwert2;
  doc["Humidity"] = sensor_BME280.Data.Messwert3;
  doc["Pressure"] = sensor_BME280.Data.Messwert4;
  doc["Time"] = timestrbuffer;
  doc["Duration"] = durationstrbuffer;
  
  doc.shrinkToFit();
  serializeJson(doc, json_str);
  // serializeJson(doc, Serial);
  // DBG__PRINT("\n\tjson_str length:", "");
  // DBG__PRINT(" ", String(measureJson(doc)));
  // DBG__PRINT("\tdoc - Memory usage:", "");
  // DBG__PRINT(" ", String(doc.memoryUsage()));
  if (id >= 0) {
    webSocket.sendTXT(id, json_str, measureJson(doc));
  }
  else {
    webSocket.broadcastTXT(json_str, measureJson(doc));
  }

}

```

bblanchon, 2024-11-28:
My review:

1. Small stack buffers like `timestrbuffer` and `durationstrbuffer` are appropriate; however, for larger buffers (let's say >128B), you should allocate on the heap to prevent stack overflows (you can also use `String` for simplicity).

2. `serializeJson()` returns the length of the JSON document, so you can use the return value in place of `measureJson()`, which will significantly improve speed and code size.

hasenradball, 2024-11-28:
Dear Benoit,

thank you so much for the review and the hints.

For the return of serialize function I was unsure, looked up in the API but was not explizitly mentioned as length of JsonDocument.

Indeed in the past I prevented as much as possible to use the heap.
I took nearly everything on stack.
Is it good putting things on heap which don‘ t last long?
An how to see a StackOverflow comming?
 
In the example below would it the also better to use std::Strings for the char arrays like timezone, password, etc.
And directly do it with std:String?
```

StaticJsonDocument<JSON_DOC_SIZE> doc;

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, file);
  if (error) {
    DBG__PRINT(F(">>> ERROR: deserializeJson() of >config.txt< failed:"), "");
    DBG__PRINT(error.f_str(), "!\n");
    return false;
  }
  else {
    // Ausgabe => Serial
    //serializeJson(doc, Serial);
    DBG__PRINT("\n");
  }

  // Copy values from the JsonDocument to the Config
  strlcpy(wlan_host, doc["wifi"]["Hostname"], sizeof(wlan_host));
  strlcpy(wlan_ssid, doc["wifi"]["SSID"], sizeof(wlan_ssid));
  strlcpy(wlan_passwd, doc["wifi"]["Password"], sizeof(wlan_passwd));
  strlcpy(time_zone, doc["time"]["TimeZone"], sizeof(time_zone));
  strlcpy(time_server1, doc["time"]["TimeServer1"], sizeof(time_server1));
  strlcpy(time_server2, doc["time"]["TimeServer2"], sizeof(time_server2));
  strlcpy(time_server3, doc["time"]["TimeServer3"], sizeof(time_server3));
  strlcpy(time_zone, doc["time"]["TimeZone"], sizeof(time_zone));
  licht_schwelle = doc["licht"]["Schwellwert"];
  licht_min = doc["licht"]["Min"];
  licht_max = doc["licht"]["Max"];
  farbe_r = doc["farbe"]["RGB-R"];
  farbe_g = doc["farbe"]["RGB-G"];
  farbe_b = doc["farbe"]["RGB-B"];
  sprach_einstellung = doc["sprache"]["Wert"];

  tim_wlan.active = doc["timerWLAN"]["active"];
  strlcpy(tim_wlan.on_time, doc["timerWLAN"]["On"], sizeof(tim_wlan.on_time));
  setTimer_struct(tim_wlan.on_time, &tim_wlan.on_time_struct);
  strlcpy(tim_wlan.off_time, doc["timerWLAN"]["Off"], sizeof(tim_wlan.off_time));
  setTimer_struct(tim_wlan.off_time, &tim_wlan.off_time_struct);
  return true;
}
```
Can you please put me on the list for the book?

bblanchon, 2024-12-02:
Stack memory is still preferred but limited to 4096 bytes on ESP8266, so it's unsuitable for large buffers.

closed 2024-12-04 by hasenradball