bblanchon/ArduinoJson #2143: Crash with build_type = debug in PlatformIO version 7.2 and later with ESP8266

WG73, 2024-11-27:
Hello,
The issue has returned. When setting build_type = debug, the ESP8266 crashes at runtime.

The problem occurs starting from version 7.2. It works fine with version 7.1.

Attached, you’ll find the decoded crash report generated using esp8266_exception_decoder.

Best regards,

**Environment**  
* Microcontroller: ESP8266
* IDE:VSCode with PlatformIO
* PainLess Mesh

[crash.txt](https://github.com/user-attachments/files/17936110/crash.txt)

WG73, 2024-11-29:
Hello Benoit,

Thank you for your reply.

The crash occurs when the PainlessMesh library establishes a connection and processes a message using ArduinoJson. This specific combination seems to trigger the issue.

I have now created a demo using PlatformIO in VSCode. When build_type is set to debug, the program crashes. Without debug, it runs without issues.

Here the code
[BugDemo.zip](https://github.com/user-attachments/files/17954860/BugDemo.zip)

Let me know how I should proceed or if you need additional information.

Best regards,

> Which issue?
> 
https://arduinojson.org/v6/issues/esp8266-crashes-in-debug-mode/

bblanchon, 2024-11-29:
The ticket in PainlessMesh mentions a failed assertion:

> src/ArduinoJson/Collection/CollectionImpl.hpp:21 void ArduinoJson::V720PB22::detail::CollectionIterator::next(const ArduinoJson::V720PB22::detail::Resou: Assertion 'currentId_ != NULL_SLOT' failed.

This probably means that the stack is corrupted or a pointer is dangling.
I can't tell without reproducing the problem locally.
Could you give me complete instructions to reproduce the bug?

Note that the failed assertion doesn't appear on your `crash.txt`, so it could also be a different issue.

For now, I don't have any reason to believe this bug is related to #1210 or #1314.

WG73, 2024-11-29:
> This probably means that the stack is corrupted or a pointer is dangling. I can't tell without reproducing the problem locally. Could you give me complete instructions to reproduce the bug?

1. user VSCODE with PlatformIO
2. create a new project (esp8266)
3.  add this to polatformio.ini
`test = aaaa
sdsd=sds`

> Note that the failed assertion doesn't appear on your `crash.txt`, so it could also be a different issue.
> 
> For now, I don't have any reason to believe this bug is related to #1210 or #1314.

WG73, 2024-11-29:
**the Exception:**

*___ Exception (2):
epc1=0x3ffffb2c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffffb2c depc=0x00000000
InstructionFetchError: Processor internal physical address or data error during instruction fetch*

>Could you give me complete instructions to reproduce the bug?

1. user VSCODE with PlatformIO
2. create a new project (esp8266)
3. add this to platformio.ini(change the com ports)

```
[env]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps = 
	painlessmesh/painlessMesh@^1.5.0	
upload_speed = 921600
monitor_speed = 115200
monitor_dtr = 0
monitor_rts = 0
monitor_filters = esp8266_exception_decoder
build_type = debug

[env:device_1] ; first device
upload_port = COM6
monitor_port = COM6

[env:device_2] ; second device
upload_port = COM19
monitor_port = COM19
```
4. add this to main.cpp
```
#include <Arduino.h>
#include <painlessMesh.h>

#define   MESH_PREFIX     "BUG_TEST_WIFI"
#define   MESH_PASSWORD   "iwillbeback"
#define   MESH_PORT       5555
#define   WIFI_CHANNEL	  13

painlessMesh painMesh;
Scheduler TaskScheduler;
uint8_t known_nodes = 0;

void receivedData(uint32_t from, String &msg);
void connect();
void newConnection(uint32_t nodeId);
void dropConnection(uint32_t nodeId);
void changedConnection();

void setup() {
	Serial.begin(115200);
	Serial.println("Start");
	connect();
	
}

void loop() {
	painMesh.update();
}

void receivedData(uint32_t from, String &msg) {
    Serial.print("Received from node ");
    Serial.print(from);
    Serial.print(": ");
    Serial.println(msg);
}

void connect() {
    WiFi.setAutoReconnect(false);
    WiFi.persistent(false);

	painMesh.setDebugMsgTypes( ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE ); // all types on

    painMesh.setRoot(true);
    painMesh.setContainsRoot(true);
	painMesh.init( MESH_PREFIX, MESH_PASSWORD, &TaskScheduler, MESH_PORT, WIFI_AP_STA, WIFI_CHANNEL);
	painMesh.onReceive(&receivedData);
	painMesh.onNewConnection(&newConnection);
	painMesh.onChangedConnections(&changedConnection);
	painMesh.onDroppedConnection(&dropConnection);	

}

void newConnection(uint32_t nodeId) {
    Serial.printf(PSTR("--> mesh::newConnection: New Connection, nodeId = %u\n"), nodeId);
}

void dropConnection(uint32_t nodeId) {
    Serial.printf(PSTR("--> mesh::dropConnection: Drop Connection, nodeId = %u\n"), nodeId);    
}

void changedConnection() {
	Serial.printf(PSTR("mesh::changedConnection-> Know Node Before:%d\n"), known_nodes);	
	Serial.printf(PSTR("mesh::changedConnection-> Know Nodes:%d\n"), known_nodes);

}
```

5. compile, upload to two devices 
6. open serial monitor
6. wait for connection
7. -> crash. here the complete [crash_complete.txt](https://github.com/user-attachments/files/17956267/crash_complete.txt)

bblanchon, 2024-11-30:
I managed to reproduce the crash thanks to your detailed instructions.

Unfortunately, I was not able to narrow the location of the bug.
When I added traces to the libraries, I saw changes in the crash behavior.
This is typical of a stack corruption, which could be caused by any library.
The fact that ArduinoJson appears (sometimes) on the stack trace doesn't prove that it is guilty.

I can't afford to spend more time on this issue.
If the error comes from ArduinoJson, then there must be a way to reproduce it without any external library.
Until someone provides this proof, I assume that ArduinoJson is not to blame.

bblanchon, 2024-11-30:
Using `git bisect`, I found that e297932a987087fa9d6732029949503650335057 is the first commit to crash the device.

bblanchon, 2024-11-30:
The crash occurs in [`painlessmesh/router.hpp` line 138](https://gitlab.com/painlessMesh/painlessMesh/-/blob/develop/src/painlessmesh/router.hpp#L138), when the `Variant` instance is destroyed.
Curiously, replacing `make_shared` (lines [110](https://gitlab.com/painlessMesh/painlessMesh/-/blob/develop/src/painlessmesh/router.hpp#L110) and [119](https://gitlab.com/painlessMesh/painlessMesh/-/blob/develop/src/painlessmesh/router.hpp#L119)) with `make_unique` solves the issue.

bblanchon, 2024-12-02:
The copy constructor of `painlessmesh`'s `Variant` class blindly copies the `jsonObj` member instead of getting a new one from `jsonBuffer`.
Unfortunately, fixing this bug doesn't fix the crash.

closed 2024-12-22 by bblanchon