bblanchon/ArduinoJson #2144: Can't use a variadic template function to *insert* new objects into document structure

mathemaphysics, 2024-12-03:
**Describe the issue**  

I want to use your variadic template function example which accesses JSON paths like jsonDocument["firstKey"]["sercondKey"]["thirdKey"] via e.g. `value<JsonArray>("firstKey", "secondKey", "thirdKey")`, but with non-const `JsonVariant` which I can modify.

Your example, which I mostly duplicated is here:  https://github.com/bblanchon/ArduinoJson/issues/1505#issuecomment-782825946

I've used perfect forwarding and it still doesn't update the document. The only method that works is direct access of the document like:

```cpp
// Add a new JsonArray at a key

template <typename TKey>
JsonVariant resolvePath(JsonVariant&& variant_, TKey&& key_)
{
    return std::forward<JsonVariant>(std::forward<JsonVariant>(variant_)[std::forward<TKey>(key_)]);
}

template <typename TKey, typename... TTrail>
JsonVariant resolvePath(JsonVariant&& variant_, TKey&& key_, TTrail&&... trail_)
{
    return std::forward<JsonVariant>(resolvePath(std::forward<JsonVariant>(std::forward<JsonVariant>(variant_)[std::forward<TKey>(key_)] | JsonObject()), std::forward<TTrail>(trail_)...));
}

template <typename... TKeys>
JsonVariant get(TKeys&&... keys_)
{
    return std::forward<JsonVariant>(resolvePath(std::forward<JsonVariant>(_data), std::forward<TKeys>(keys_)...));
}

// This works
JsonArray newArrayRef = jsonDocument["firstKey"]["secondKey"]["thirdKey"].to<JsonArray>();

// This doesn't work
JsonArray newArrayRef = get(jsonDocument, "firstKey", "secondKey", "thirdKey").to<JsonArray>();

// Fill out the new JsonObject created inside
JsonObject newElement = newArrayRef.add<JsonObject>();
newElement["a"] = "someValue";
newElement["b"] = "someOtherValue";
```

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v7/troubleshooter/):  

This doesn't match the problem I have, but here is the closest I could find.

[image]

**Environment**  

Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: Using PlatformIO
* IDE: VSCode with PlatformIO extension

**Reproduction**  
Here is a small snippet that demonstrate the problem.

I used your example in https://github.com/bblanchon/ArduinoJson/issues/1505#issuecomment-782825946, but I replaced the `JsonVariantConst` types with `JsonVariant`. I want to be able to insert new objects using the same methodology. This doesn't work. It only works when I insert using the chain of `operator[]` calls followed by a `.template to<JsonArray>()`.

To summarize: I want to use your `value(jsonDocument, "firstKey", "secondKey", "thirdKey").to<JsonArray>()` (using ArduinoJson version 7 methodology, which I read is to create with `to`).

This categorically doesn't work. I've tried perfect forwarding in the variadic template functions for all functions involved. It still won't actually modify the `JsonVariant` (non-const) which is returned by `value`.

I'm assuming in the recursive calls to templates, it fails to create the intermediate objects along the `operator[]` chain.

How would I go about writing a variadic template to create a key like `jsonDocument["firstKey"]["secondKey"]["thirdKey"]` when even `jsonDocument["firstKey"]["secondKey"]` may not exist? Do I need to manually create them step by step?

mathemaphysics, 2024-12-03:
I think I may have solved this. It requires adding `if (variant_[key_].isNull()) ...` then use `variant_[key_].template to<JsonVariant>()` else use `variant_[key_]` or `variant_[key_].template as<JsonVariant>`.

bblanchon, 2024-12-04:
Hi @mathemaphysics,

If you can use C++14, you can try the following:

```c++
template <typename Variant, typename Key>
decltype(auto) resolvePath(Variant&& variant, Key&& key) {
  return variant[std::forward<Key>(key)];
}

template <typename Variant, typename Key, typename... Tail>
decltype(auto) resolvePath(Variant&& variant, Key&& key, Tail&&... tail) {
  return resolvePath(variant[std::forward<Key>(key)], std::forward<Tail>(tail)...);
}
```

Online demo: https://wandbox.org/permlink/KsfKvorK7HNcJY83

Best regards,
Benoit

mathemaphysics, 2024-12-04:
Thanks @bblanchon.

I didn't think of trying that. I'll give it a try.

mathemaphysics, 2024-12-06:
I tried the method you suggested, but the pre-C++14 version, i.e.

```cpp
template <typename Variant, typename TKey, typename... TTrail>
auto resolvePath(Variant&& variant_, TKey&& key_) -> decltype(variant_[std::forward<TKey>(key_)])
{
    return variant_[std::forward<TKey>(key_)];
}
```

It appears that it doesn't operate the same way the simple variant_[key1][key2][key3] does. I'm not sure why it wouldn't though. Each recursive call to `resolvePath` will produce the next element in the chain of `operator[]`.

bblanchon, 2024-12-06:
Doing the same without C++14 is too tricky; instead, you should call `to<JsonVariant>()` along the way:

```c++
JsonVariant createNestedVariant(JsonVariant variant) {
  return variant;
}

template <typename Key, typename... Tail>
JsonVariant createNestedVariant(JsonVariant variant, Key&& key, Tail&&... tail) {
  return createNestedVariant(
    variant[std::forward<Key>(key)].template to<JsonVariant>(), 
    std::forward<Tail>(tail)...
  );
}
```

Online demo: https://wandbox.org/permlink/VnfvYTQVvTIZJFgZ

mathemaphysics, 2024-12-09:
@bblanchon That is also what I concluded. I was able to get it to do everything I need it to using `to<JsonArray>`.

closed 2024-12-09 by mathemaphysics

bblanchon, 2024-12-09:
You're welcome, @mathemaphysics.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

mathemaphysics, 2024-12-12:
@bblanchon Well deserved star.

A partner and I have been using ArduinoJson as the core of a flow computer system. It's serving as the variant-centric means of passing hierarchical constructor arguments around for using in setup of connected devices.

Very nicely organized and efficient. I love it. Thanks for taking the time to put it together, and also help with troubleshooting.