bblanchon/ArduinoJson #2146: [switch to v7.2.1] Need Help for Type IPAddress

hasenradball, 2024-12-20:
Wanted to use the following code:

```
  JsonDocument doc;

  doc["version"] = VERSION;
  doc["WLAN_USER_INPUT_Requested"] = wlan_user_req;

  JsonObject wifi = doc.to<JsonObject>();
  wifi["MAC"] = WiFi.macAddress();
  wifi["IP"] = WiFi.localIP();
  wifi["Hostname"] = io.getHost();
  
  ```
  Get to following error:
  
  `.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp:33:41: error: conversion from 'const IPAddress' to 'long long int' is ambiguous`

How to solve this probperly?

hasenradball, 2024-12-20:
```
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_features.cpp.o
In file included from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/type_traits.hpp:15,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/utility.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Memory/MemoryPoolList.hpp:9,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:8,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Array/JsonArray.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.hpp:29,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.h:9,
                 from lib/cConfigIO_WordClock/cConfigIO_WordClock.h:5,
                 from lib/serverfunctions/serverfunctions.h:6,
                 from lib/serverfunctions/serverfunctions.cpp:2:
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp: In instantiation of 'const bool ArduinoJson::V721PB22::detail::is_convertible<IPAddress, long long int>::value':
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:301:30:   recursively required by substitution of 'template<class T> struct ArduinoJson::V721PB22::Converter<T, typename ArduinoJson::V721PB22::detail::enable_if<ArduinoJson::V721PB22::detail::is_enum<T>::value, void>::type> [with T = IPAddress]'
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:301:30:   required from 'bool ArduinoJson::V721PB22::detail::VariantRefBase<TDerived>::doSet(T&&) const [with TConverter = ArduinoJson::V721PB22::Converter<IPAddress, void>; T = const IPAddress&; TDerived = ArduinoJson::V721PB22::detail::MemberProxy<ArduinoJson::V721PB22::JsonObject, const char*>]'
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:80:44:   required from 'bool ArduinoJson::V721PB22::detail::VariantRefBase<TDerived>::set(const T&) const [with T = IPAddress; TDerived = ArduinoJson::V721PB22::detail::MemberProxy<ArduinoJson::V721PB22::JsonObject, const char*>]'
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:33:14:   required from 'ArduinoJson::V721PB22::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >& ArduinoJson::V721PB22::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >::operator=(const T&) [with T = IPAddress; TUpstream = ArduinoJson::V721PB22::JsonObject; TStringRef = const char*]'
lib/serverfunctions/serverfunctions.cpp:134:29:   required from here
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp:33:41: error: conversion from 'const IPAddress' to 'long long int' is ambiguous
   33 |   static const bool value = sizeof(probe(from_)) == sizeof(int);
      |                                   ~~~~~~^~~~~~~~
In file included from /home/frank/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from /home/frank/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30,
                 from lib/serverfunctions/serverfunctions.h:5,
                 from lib/serverfunctions/serverfunctions.cpp:2:
/home/frank/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/IPAddress.h:80:9: note: candidate: 'IPAddress::operator uint32_t() const'
   80 |         operator uint32_t() const { return isV4()? v4(): (uint32_t)0; }
      |         ^~~~~~~~
/home/frank/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/IPAddress.h:84:9: note: candidate: 'IPAddress::operator bool() const'
   84 |         operator bool () const { return isSet(); } // <-
      |         ^~~~~~~~
In file included from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/type_traits.hpp:15,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Polyfills/utility.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Memory/MemoryPoolList.hpp:9,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:8,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/JsonVariantConst.hpp:10,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:9,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Array/ElementProxy.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Array/JsonArray.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.hpp:29,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.h:9,
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_flash_quirks.cpp.o
                 from lib/cConfigIO_WordClock/cConfigIO_WordClock.h:5,
                 from lib/serverfunctions/serverfunctions.h:6,
                 from lib/serverfunctions/serverfunctions.cpp:2:
[... cut ...]
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_non32xfer.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_noniso.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_phy.cpp.o
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_postmortem.cpp.o
*** [.pio/build/d1_mini/lib6be/serverfunctions/serverfunctions.cpp.o] Error 1
================================================================ [FAILED] Took 5.63 seconds ================================================================

```

hasenradball, 2024-12-20:
Is this sufficient?

`  wifi["IP"] = WiFi.localIP().toString();`

bblanchon, 2024-12-22:
Hi @hasenradball,

Calling `toString()` is the simplest solution, but you could also create [a custom converter](https://arduinojson.org/v7/how-to/create-converters-for-ipaddress/).

Best regards,
Benoit

hasenradball, 2024-12-22:
Hi Benoit,

thank you and Merry Christmas!
Anny update about the book?

bblanchon, 2024-12-23:
You're welcome. Merry Christmas to you, too.
Unfortunately, I currently don't have the time to work on the paperback version of my book.

closed 2024-12-23 by bblanchon