bblanchon/ArduinoJson #2147: Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)

sanluca, 2024-12-22:
Hi,

I'm having trouble parsing JSON data using ArduinoJson on my ESP32. I'm encountering several errors during debugging (see attached). The code I'm using is also included. Any help would be greatly appreciated. Thanks.

[debug.txt](https://github.com/user-attachments/files/18221159/debug.txt)

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: 3.0.7
* IDE: 2.3.4

[mqttbridge.zip](https://github.com/user-attachments/files/18221163/mqttbridge.zip)

bblanchon, 2024-12-23:
Hi @sanluca,

Please use the [ArduinoJson Troubleshooter](https://arduinojson.org/troubleshooter/) and share the report.
If the Troubleshooter doesn't help you find the solution, provide an [MCVE](https://www.stackoverflow.com/help/mcve) with a maximum of 40 lines and no third-party library.

Best regards,
Benoit

sanluca, 2025-01-04:
I tried version 7.3.0 and so far it hasn't given that error again. I'm investigating

thanks

closed 2025-01-13 by bblanchon