bblanchon/ArduinoJson #2148: (feat) JsonVariant.<T>orElse(T default)

mathieucarbou, 2024-12-24:
Hello,

I didn't see a method that would be the equivalent as checking if a json variant is there, and if not, return a default value provided by the user, but not by the library.

In my case, I want floats to be set to NAN if the value is not there in the Json.

Example of what I have to write now:

```c++
        JsonObject aggregate = doc["aggregate"].as<JsonObject>();
        grid.remoteMetrics().update({
          .apparentPower = aggregate.containsKey("apparent_power") ? aggregate["apparent_power"].as<float>() : NAN,
          [...]
        });
```

What I would like to write:

```c++
.apparentPower = aggregate["apparent_power"].orElse<float>(NAN),
```

I didn't see such thing in `VariantRefBase.hpp` ?

Happy Christmas ! 🎄

bblanchon, 2024-12-26:
Hi @mathieucarbou,

You can use [operator |](https://arduinojson.org/v7/api/jsonvariant/or/), like so:

```c++
.apparentPower = aggregate["apparent_power"] | NAN,
```

Best regards,
Benoit

mathieucarbou, 2024-12-26:
Wow, nice! I didn't saw it in the jsonvariant doc :-) thanks!

closed 2024-12-26 by mathieucarbou