bblanchon/ArduinoJson #2150: String comparison yields wrong result

avispeng, 2025-01-08:
**Describe the bug**  
If I store a string in a JSON document, extract the same string (`const char*`) from the document, and compare it with another string using `strcmp` or `strncmp`, the result might be wrong. To be specific, if the first string is a prefix of the second string, the result incorrectly yields 0, i.e. two strings are identical.

**Troubleshooter report**  
The issue is not related to serializing or deserializing thus there is no report.

**Environment**  
Here is the environment that I used:
* Microcontroller: STM32F767ZI
* Core/runtime: Arduino Core for STM32 2.2.0
* IDE: PlatformIO Core, version 6.1.16

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
StaticJsonDocument<128> doc;
doc["str1"] = "0.0";
const char* str1 = doc["str1"].as<const char*>();
const char* str2 = "0.0.0";
if (strcmp(str1, str2) == 0) {
  Serial.println("strcmp: str1 equals str2");
}
```

**Program output**  

Expected output:
The serial should not print anything because str1 and str2 are not equal

Actual output:

```
strcmp: str1 equals str2
```

**Other information**
I found that Arduino String has this same issue when not using ArduinoJson.
```c++
String str3 = "V";
Serial.println(str3.equals("VI")); // this should print 0 (false) but it printed 1 (true)
```

bblanchon, 2025-01-09:
Hi @avispeng,

You didn't specify the ArduinoJson version, so I tried with the last one (7.3.0), but could not reproduce the issue.
See an online demo here: https://wandbox.org/permlink/EZz08VrCJBaOZjD2

I also tested with the last version on the v6 branch (6.21.5) and couldn't reproduce either.
See an online demo here: https://wandbox.org/permlink/mi0NQSdw73iCJQv3

> I found that Arduino String has this same issue when not using ArduinoJson.

This clue indicates that the problem is not related to ArduinoJson.
I'd recommend uninstalling and reinstalling the Arduino Core for STM32.
By the way, do you get the same issue with other cores?

Best regards,
Benoit

avispeng, 2025-01-09:
Hi Benoit,

Thank you for the reply. I tried with ArduinoJson v6 and v7 and got the same result. But you are right, this issue is probably related to the core. I wasn't using the latest Arduino Core for STM32. I tried to update to the latest today but I had other weird problems with Serial connection thus I couldn't check if it got fixed with the updated core. I didn't have the chance to try other cores either. Anyway, I will keep digging. Thank you again.

Best,
H. Peng

bblanchon, 2025-01-10:
You're welcome.
Please let us know how you fix the issue in the end.

closed 2025-01-10 by bblanchon

avispeng, 2025-01-10:
> You're welcome. Please let us know how you fix the issue in the end.

I updated the core to the latest (framework-arduinoststm32@4.20900.0 in PlatformIO, which is Arduino_Core_STM32@2.9.0), and also the toolchain-gccarmnoneeabi in platformio (the GNU Arm Embedded Toolchain, where `strcmp` is from). The issue persisted.

But I finally located the issue. It is not related to the core, or the toolchain, it is related to wolfboot, which is used in our firmware. I made the minimal firmware for debugging by removing everything else from my main.cpp, but I forgot the line that includes wolfboot.h (my bad, really sorry). When not including it, the string comparison yields correct result. It turns out, wolfboot ovewrites implementation of `strcmp` (https://github.com/wolfSSL/wolfBoot/blob/master/src/string.c#L106), and in their implementation, the comparison ends when s1 reaches the end, and return the diff "so far", if s1 is a prefix of s2, when s1 ends, there is no difference..

Anyway, I am glad this was identified finally. Thank you for your time. Have a nice day!