bblanchon/ArduinoJson #2153: JSON Deserialization

cmcooper1971, 2025-01-25:
**Description**
Please see the code, which is called and pulls  in chart data from a Yahoo API data provider. It works OK for the first few ticker symbols, but then errors.

The error depends on how I am trying to get the data, as I have tried your online assistant and all of its recommendations, including:

Using a String
Using LittleFS
Using a Stream
Using StreamUtils

At present from the code, you will see that I have reverted to using a String.

This either errors as "EmptyInput" or "NoMemory"

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns deserialization
4. `deserializeJson()` returns `EmptyInput`
5. Input comes from an HTTP response
6. The status code is not in the list
7. Increasing the timeout doesn't fix the issue
8. The input is a `String`
9. Passing a `Stream` to `deserializeJson()` doesn't fix the issue

**Environment**
* Microcontroller: ESP32 DEV Kit C
* Core/Framework: ESP32 on Arduino IDE 2
* IDE: Arduino IDE 2

**Reproduction code**
```c++
/*-----------------------------------------------------------------*/

// Function to fetch and parse JSON chart data from Yahoo Finance API

void fetchChartData(const String& symbol) {

	if (WiFi.status() == WL_CONNECTED) {

		HTTPClient http;

		// Construct the URL with the passed symbol
		String url = "https://yahoo-finance-api-data.p.rapidapi.com/chart/simple-chart?symbol=" + symbol + "&limit=300&range=1y";

		http.begin(url);
		http.addHeader("x-rapidapi-key", apiKey);
		http.addHeader("x-rapidapi-host", "yahoo-finance-api-data.p.rapidapi.com");

		// Increase timeout
		http.setTimeout(10000); // Wait up to 10 seconds for the server response

		int httpCode = http.GET();

		Serial.print("Chart HTTP Code :");
		Serial.println(httpCode);

		if (httpCode > 0) {

			String payloadChart = http.getString(); // Retrieve data

			Serial.println("Chart HTTP Payload: OK");
			outputDebug("HTTP Payload: ");
			outputDebugLn(payloadChart);

			http.end();

			// Define JSON filter
			StaticJsonDocument<200> filter;
			filter["data"][0]["meta"]["range"] = true;
			filter["data"][0]["timestamp"] = true;
			//filter["data"][0]["indicators"]["quote"][0]["open"] = true;
			filter["data"][0]["indicators"]["quote"][0]["close"] = true;
			//filter["data"][0]["indicators"]["quote"][0]["high"] = true;
			//filter["data"][0]["indicators"]["quote"][0]["low"] = true;
			//filter["data"][0]["indicators"]["quote"][0]["volume"] = true;

			// Create a document with reduced memory capacity
			DynamicJsonDocument docChart(35000);

			// Deserialize with the filter
			DeserializationError error = deserializeJson(docChart, payloadChart, DeserializationOption::Filter(filter));

			if (!error) {
				JsonObject meta = docChart["data"][0]["meta"];
				JsonArray timestamps = docChart["data"][0]["timestamp"];
				//JsonArray opens = docChart["data"][0]["indicators"]["quote"][0]["open"];
				JsonArray closes = docChart["data"][0]["indicators"]["quote"][0]["close"];
				//JsonArray highs = docChart["data"][0]["indicators"]["quote"][0]["high"];
				//JsonArray lows = docChart["data"][0]["indicators"]["quote"][0]["low"];
				//JsonArray volumes = docChart["data"][0]["indicators"]["quote"][0]["volume"];

				// Parse and store chart range
				String chartRange = meta["range"];
				chartRangeT = chartRange;
				Serial.println("");
				Serial.println("Chart Range: " + chartRange);
				Serial.println("");

				// Parse and store chart data
				instanceCount = timestamps.size();
				Serial.println("Original Number of Instances: " + String(instanceCount));
				Serial.println("");

				size_t validInstanceCount = 0;
				for (size_t i = 0; i < timestamps.size(); i++) {
					//float openValue = opens[i].isNull() ? 0.0 : opens[i].as<float>();
					float closeValue = closes[i].isNull() ? 0.0 : closes[i].as<float>();
					//float highValue = highs[i].isNull() ? 0.0 : highs[i].as<float>();
					//float lowValue = lows[i].isNull() ? 0.0 : lows[i].as<float>();
					//long volumeValue = volumes[i].isNull() ? 0 : volumes[i].as<long>();

					//if (openValue == 0.0 && closeValue == 0.0 && highValue == 0.0 && lowValue == 0.0 && volumeValue == 0) {
					//	continue;
					//}

					if (closeValue == 0.0) {
						continue;
					}

					timestampsArray[validInstanceCount] = timestamps[i].as<long>();
					//opensArray[validInstanceCount] = openValue;
					closesArray[validInstanceCount] = closeValue;
					//highsArray[validInstanceCount] = highValue;
					//lowsArray[validInstanceCount] = lowValue;
					//volumesArray[validInstanceCount] = volumeValue;

					validInstanceCount++;
				}

				instanceCount = validInstanceCount;
				Serial.println("Updated Number of Instances: " + String(instanceCount));
				Serial.println("");

				// Clear String

				payloadChart = "";

				// Clear docChart

				docChart.clear();

			}

			else {
				Serial.print("Chart JSON Parsing Error: ");
				Serial.println(error.c_str());
			}

		}

		else {
			Serial.print("Chart HTTP Request failed: ");
			Serial.println(httpCode);
		}

		http.end();
	}

	else {
		Serial.println("WiFi not connected");
	}

} // Close function

/*-----------------------------------------------------------------*/
```

**Remarks**
I love your library, I have used it previously for a Weather Station using Open Weather (please see it here - https://youtu.be/gPtN3tGBm78?si=B5zkXfrhU6xy1zKp), but I am building a new project which uses the Weather Station hardware, but being repurposed as a Stock Ticker display board. I fear that the size of the API call is simply too much for the ESP32!

bblanchon, 2025-01-27:
Hi @cmcooper1971,

I'm assuming you get the following:

1. `EmptyInput` with `getString()`
2. `NoMemory` with `getStream()`

Is this correct?
If so, the response is too big for the microcontroller, and you must stick to `getStream()`.
See [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v7/how-to/use-arduinojson-with-httpclient/)

Using a filter, as you did, usually helps fit the result in the `JsonDocument`.
Please check with the [ArduinoJson Assistant](https://arduinojson.org/v7/assistant) that there is enough RAM after filtering the response.

Also, please tell us what you see in the Serial Monitor.

BTW, you said to the [Troubleshooter](https://arduinojson.org/troubleshooter/) that you use AduinoJson 7, but the code uses ArduinoJson 6 syntax. What's up with that?

Best regards,
Benoit

cmcooper1971, 2025-01-27:
Hi @bblanchon,

Thank you for replying to my question.

For information, I am using the LVGL library for the TFT display and my ESP partition scheme is set to Huge, 3Mb, NoOTA and 1MB Spiffs.

Answering your final point first, I have tried v5, v6 & v7 and many iterations, but in my latest function it is as it is supposed to be as follows:

```
// Allocate JsonBuffer
JsonDocument docChart;

//Deserialize with the filter
DeserializationError error = deserializeJson(docChart, payloadChart, DeserializationOption::Filter(filter));
```

Returning to your initial question, when using a `getString()` I receive the error `NoMemory` after the function has been called a number of times. 

When I swap to `getStream()` I receive the error `IncompleteInput`, once again after the function has been called a number of times. 

As this looks to be a buffering problem, I then use the `StreamUtils` library you reference in your online Assistant.

When I call a 6 month chart, it includes ~125 entries, while a 1 year chart includes ~250 entries. If I use 6 months, the function works for a while, but when I choose 1 year, I receive `IncompleteInput` straight away.

If it is a lack of memory, then in my view, it should never work which leads me to think it must be a memory stack issue. So previously I switched to using LittleFS for the String data but that didn't make any difference.

When considering your assistant says my JSON query uses 28k peek, I struggle to understand how or why it should be a memory issue as when compiled as follows:

[image]

`Program Stock_Market_Watch_LVGL size: 1511569 bytes (used 48% of a 3145728 byte maximum) (62.89 secs)
Minimum Memory Usage: 113848 bytes (35% of a 327680 byte maximum)`

Any advice would be appreciated.

Regards,

Christopher

bblanchon, 2025-01-29:
> after the function has been called a number of times

This sounds like a memory leak or [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html).
Monitor `ESP.getFreeHeap()` and `ESP.getMaxAllocHeap()` over time to see which one it is.

If `ESP.getFreeHeap()` goes down, it's a memory leak, so hunt it down.

If `ESP.getMaxAllocHeap()` goes down, it's a case of heap fragmentation. Therefore, reduce your use of heap-allocated variables, such as `String`.
ArduinoJson 7 is designed to produce little fragmentation, but if in doubt, you can switch back to ArduinoJson 6, which shouldn't produce any fragmentation.

cmcooper1971, 2025-01-30:
Thank you once again.

I will perform some testing and will advise an outcomes.

cmcooper1971, 2025-02-01:
I have to start with an apology for wasting your time.

It seems the LVGL library was causing memory leeks in the functions I had written, which only manifested themselves when I then implemented your JSON library to obtain the data.

I am now obtaining lots of financial data, either using a String or Stream without any issues time and time again.

Thank you once again for your time and of course your library!

closed 2025-02-01 by cmcooper1971