bblanchon/ArduinoJson #2156: Error compiling with Arduino Cli 1.1.1

wilsondzz, 2025-01-30:
**Description**
Working with "Arduino-Cli" being run from VSC, no Arduino IDE installed.

I used the "Arduino:Library Manager" to install the ArduinoJSON Library.

ArduinoJson by Benoit Blanchon <http://blog.benoitblanchon.fr/> Version 7.3.0 INSTALLED
An efficient and elegant JSON library for Arduino Supports JSON parsing and formatting. Uses fixed 
memory allocation.

During compile:

from D:\Arduino\2025\Internet_Status_Dome\Internet_Status_Dome.ino:2:
#include <Arduino.h>
#include <ArduinoJson.h>
#include <ArduinoMqttClient.h>

C:\Users\redacted\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp:27:20: note:   initializing argument 1 of 'static int ArduinoJson::V730PB22::detail::is_convertible<From, To>::probe(To) [with From = IPAddress; To = long long int]'
27 |   static int probe(To);
|                    ^~
Error during build: exit status 1

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. The error is not in the list

**Environment**
* Microcontroller: esp8266:esp8266:nodemcuv2

**Reproduction code**
```c++
C:\Users\redacted\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp:27:20: note:   initializing argument 1 of 'static int ArduinoJson::V730PB22::detail::is_convertible<From, To>::probe(To) [with From = IPAddress; To = long long int]'
27 |   static int probe(To);
|                    ^~
Error during build: exit status 1
```

**Remarks**
see description.

bblanchon, 2025-02-03:
Hi @wilsondzz,

> note: initializing argument...

A line that starts with `note:` is not a compilation error but additional information about an error.
Scroll up a few lines to locate the message starting with `error:`, then ask the troubleshooter about this error.

Best regards,
Benoit

wilsondzz, 2025-02-03:
I found the error:

D:\Arduino\2025\SKETCH\SKETCH.ino:**172**:38:   required from here
C:\Users\deadbeef\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/type_traits/is_convertible.hpp:33:41: error: conversion from 'const IPAddress' to 'long long int' is ambiguous
33 |   static const bool value = sizeof(probe(from_)) == sizeof(int);
|                                   ~~~~~~^~~~~~~~

then looked at line 172, and I see the ACTUAL error:

doc["ip"]     = **WiFi.localIP()**;

should be:

doc["ip"]     = **WiFi.localIP().toString()**;

Thanks for the help, I looked at the other "Arduino_Json" lib and was trying to use it, but, I have got so use to this JSON library and really appreciate the work that you put into it, very nice.

see you around.

closed 2025-02-03 by wilsondzz