bblanchon/ArduinoJson #2158: Arduino v3.1.2 String crashes on serializeJson

Tobi177, 2025-02-15:
**Describe the bug**  
On the latest Arduino release v3.1.2 the function serializeJson does not work anymore and the app is crashing instantly. Within the Arduino release I see that some changes to the String objects were done: https://github.com/espressif/arduino-esp32/issues/10938 Maybe, around this issue is the reason for the crashes, but that is just an assumption. Switching back to previous releases are working fine again. I rollback to Arduino v3.1.1 and the app is working fine again.

**Environment**  
esp-idf v5.3 and Arduino v3.1.2 as a component
ArduinoJson v7.3.0
esp32, esp32c2

**Reproduction**  

```
String getAvailableNetworks() {

  JsonDocument json_doc;
  JsonArray networks = json_doc.to<JsonArray>();

  int n = WiFi.scanNetworks(false, false);
  if (n) {
    for (int i = 0; i < n; ++i) {
      JsonObject networkObj = networks.add<JsonObject>();
      networkObj["ssid"] = WiFi.SSID(i);
      networkObj["rssi"] = WiFi.RSSI(i);
      networkObj["authmode"] = (WiFi.encryptionType(i) == WIFI_AUTH_OPEN) ? 0 : 1;
    }
  }

  String json_available_networks;
  serializeJson(json_doc, json_available_networks);  // CRASHES HERE

  return json_available_networks;

}

```

Only the String handling is affected! Once I use char arrays/pointers the serializeJson work fine on Arduino v3.1.2 as well. Only if strings are serialized the problem occurs. Please report back if any further information are needed. Thank you.

bblanchon, 2025-02-17:
Hi @Tobi177,

Thank you for reporting this issue.

This issue was reported in espressif/arduino-esp32#10971 and fixed in espressif/arduino-esp32#10972.
Please confirm that this PR fixes the crash.

Best regards,
Benoit

Tobi177, 2025-02-21:
Thank you for your reply!

I have tested the latest Arduino release v3.1.3 which includes the mentioned PR. The problem is fixed.

Thank you!

bblanchon, 2025-02-24:
You're welcome, @Tobi177.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2025-02-24 by bblanchon