bblanchon/ArduinoJson #2164: Issue regarding ElementProxy in ArduinoJSON libs

GameKnightt, 2025-03-18:
**Describe the issue**  

Hello,

I have a project where I need to communicate information via the serial link of my Arduino Mega 2560 using JSON to send it to my Node-Red panel. I installed ArduinoJSON version 7.3.0 but when I compile I get an error about what seems to be called "ElementProxy" (see the output of my terminal). Looking at the library files (here: ArduinoJson/Array/...) there is no file "ElementProxy.hpp". Have you ever had this error? Thank you in advance for your help.

**Troubleshooter report**  
No solutions.

**Environment**  
Here is the environment that I'm using':
* Microconroller: Arduino Mega 2560 (megaatmega2560)
* Core/runtime: Mega 2560 core for PlatformIO V6.1.18
* IDE: Visual Studio Code V1.98.2
* Libs : ArduinoJSON V7.3.0

Actual output:

```
compilation terminated.
In file included from .pio\libdeps\megaatmega2560\ArduinoJson\src/ArduinoJson.hpp:29:0,
                 from .pio\libdeps\megaatmega2560\ArduinoJson\src/ArduinoJson.h:9,
                 from src\AxisDriver\SerialJSON.h:5,
                 from src\AxisDriver\MACHINE.h:8,
                 from src\AxisDriver\main.cpp:6:
.pio\libdeps\megaatmega2560\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7:10: fatal error: ArduinoJson/Array/ElementProxy.hpp: No such file or directory
 #include <ArduinoJson/Array/ElementProxy.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\megaatmega2560\src\AxisDriver\MACHINE.cpp.o] Error 1
*** [.pio\build\megaatmega2560\src\AxisDriver\SerialJSON.cpp.o] Error 1
*** [.pio\build\megaatmega2560\src\AxisDriver\main.cpp.o] Error 1
```

Thanks for your help !

bblanchon, 2025-03-18:
@GameKnightt, how did you solve your issue?

GameKnightt, 2025-03-19:
@bblanchon 

This is a project I'm doing in my school's infrastructure and I realized while downloading the library that some files were missing because the school's proxy was blocking the installation of some files. So I connected to a private network and everything worked fine.