bblanchon/ArduinoJson #2165: deserializeJson of String to JsonDocument restarts ESP8266 with WDT reason

milanbx, 2025-03-22:
**Description**
I am running ESP8266 ESP-01 with Blynk lib and there is a webserver endpoint named "vxy", in which deserialization of JSON payload is called. And for some reasen, if it comes to the deserializeJson() call, ESP restarts from a WDT reason.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns deserialization
4. The program crashes
5. Program crashes inside `deserializeJson()`
6. Removing dangling pointers doesn't prevent the crash
7. Reducing stack usage doesn't prevent the crash

**Environment**
* Microcontroller: ESP8266 ESP-01
* Core/Framework: ESP8266 Core for Arduino v3.1.2
* IDE: Arduino CLI v1.1.1

**Reproduction code**
```c++
The webserver endpoint handler setup:
  webServer.on("/vxy", HTTP_POST, VXYendpoint);

And the request handler contains following:
void VXYendpoint()
{
  if (webServer.hasArg(PLAIN))
  {
    webServer.send(HTTP_CODE_OK);
    String body(webServer.arg(PLAIN));
    JsonDocument jsonDoc;
    DeserializationError err = deserializeJson(jsonDoc, body);
    if !(err)
    {
      JsonObject root = jsonDoc.as<JsonObject>();
      for (JsonPair kv : root)
      {
        int pin = atoi(String(kv.key().c_str()).substring(1).c_str());
        const char *value = kv.value().as<const char *>();
        Blynk.virtualWrite(pin, value);
      }
    }
  }
  else // request nemá tělo
  {
    webServer.send(HTTP_CODE_BAD_REQUEST);
  }
}

```

milanbx, 2025-03-22:
As for memory management, here are runtime figures:
chip size 1048576
sketch size 384912
space 643072
heap 38952
fragmentation 2

milanbx, 2025-03-23:
I have found the cause of the WDT reset. It had no relation to **Arduino JSON** or `deserializeJson()` at all. Sorry for that.
The reason for the reset was in a fixed length `char[]` variable declartation which was not long enought to contain a string I saved in it with `strcpy()`. Good to know to avoid this by using `strncpy()` instead and control the length of the stored string that way.

closed 2025-03-23 by milanbx

bblanchon, 2025-03-24:
Thank for the explanation, @milanbx.

I added a step with the following text:

> The crash could also be caused by an [undefined behavior](https://en.wikipedia.org/wiki/Undefined_behavior) elsewhere in your program.
>
> Please comment out pieces of code until the program stops crashing.  
> Keep the call to `deserializeJson()` and the code around it.

I decided to go with something shorter to not discourage the reader, and more generic to cover UB in general.

milanbx, 2025-03-24:
Great. I though so :-)