bblanchon/ArduinoJson #2166: using const char szJSONKey[] as key

EvEggelen, 2025-04-04:
I am upgrading from ArduinoJson@7.2 to the latest version 7.3.
Code that was working before breaks. I looked at https://arduinojson.org/news/2024/12/29/arduinojson-7-3/ to see what is changed in the API.

Likely my limited knowledge, but from that text I could not find why the code below is not working anymore. Basically I have the keys as a static const char in my class to reuse the key in all the places I need it. 

```
static const char szJSONKey[]; -> in class definition

const char clas::szJSONKey[]="key";  => in the cpp file

bool function(const time_t time, JsonObject root) noexcept
{
 root[szJSONKey] = ;
 }
```

Compilation error:
error: assignment of read-only location 'Class::szJSONKey[((int)root.ArduinoJson::V731HB22::JsonObject::operator bool())]'

I just noticed that by adding  `root[JsonString(szJSONKey,true)] = 10; `makes sure it builds. I am not sure if this the most efficient method to do this. If so, I think it will help others too when adding this to your examples on : https://arduinojson.org/news/2024/12/29/arduinojson-7-3/. For me it was not clear this was also application when using const char* as keys.

Can you indicate if this is the best way of using keys with version 7.3 ?

EvEggelen, 2025-04-06:
The problem seems not only related to static const char szJSONKey[];. In a very simple application I could not reproduce it. It took a number of tests to reproduce the compilation error. 

After looking in more detail in the larger program, it seems it has something to do with using this construction in a member function in a derived class with public StateBase<CBASE>. If you do the same in the main(), it seems to work fine. 

Hope this helps.

```
#include <stdio.h>
#include <unistd.h>
#include "nvs_flash.h"
#include "esp_log.h"

#include <stdio.h>
#include <inttypes.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_log.h"
#include "driver/gpio.h"

#include "esp_timer.h"

#include <ArduinoJson.h>

static const char *TAG = "MAIN";

extern "C" {
    void app_main(void);
}

namespace test_namespace {

class CCLASS
{
public:
	static const char mszKey[];
};

class CBASE
{
public:
	static const char mszKeyb[];
};

template <class T>
class StateBase 
{
	public:
	T mState;
};

class CDERIVED : public StateBase<CBASE>
{
public:
void test(JsonObject root)
{
	
	root[JsonString(CBASE::mszKeyb,true)]=13; //<--working
	root[CBASE::mszKeyb]=13; //<--not working
}
};
}

using namespace test_namespace;

const char CCLASS::mszKey[]="test3";
const char CBASE::mszKeyb[]="test4";

void testFunction(JsonObject root)
{
	const char szKey[]="Test2";

	root["test"]=10;
	root[szKey]=11;
	root[CCLASS::mszKey]=(uint16_t)12;
	root[CBASE::mszKeyb]=(uint16_t)13;
	//root[CDERIVED::mszKeyb]=(uint16_t)14;
}

void app_main(void)
{
  ESP_LOGI(TAG, "Starting");

	// Initialize NVS
	esp_err_t ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
	{
		ESP_ERROR_CHECK(nvs_flash_erase());
		ret = nvs_flash_init();
	}
	ESP_ERROR_CHECK(ret);

	//startTests();

	sleep(1);

	JsonDocument doc;
	JsonObject root = doc.to<JsonObject>();
	testFunction(root);
    

	ESP_LOGI(TAG, "Baud rate error ");

}
```

closed 2025-04-07 by bblanchon

EvEggelen, 2025-04-07:
Not problem. Thanks for all the work on ArduinoJson. I will update as soon as it becomes available.

bblanchon, 2025-04-09:
The fix is available in [ArduinoJson 7.4.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.4.0)

EvEggelen, 2025-04-09:
Included version 7.4.0 in my project and reverted all my changes. So far it seems to be problem free during my brief tests. Thanks for solving the issue.

bblanchon, 2025-04-10:
You're welcome, @EvEggelen.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

commit 7a4eee95 (2025-04-07): Fix support for `const char[]`

Fixes #2166
commit f0e84e49 (2025-04-09): Fix support for `const char[]`

Fixes #2166